OPCFoundation/UA-.NETStandard #3: There is no cert with subject UA Sample Server in the configuration.

robinson, 2016-04-26:
When I running the server sample in Windows 10, occurred this message:
There is no cert with subject UA Sample Server in the configuration.
 Copy cert to this location:
C:\Users\[user]\AppData\Local\Packages\[packageName]\LocalState\OPC Foundation\CertificateStores\MachineDefault

Althought I copied the .cert file which I generated with subject name: UA Sample Server.

robinson, 2016-04-26:
So, when I created a new folder named: certs, then it is running.

closed 2016-04-26 by robinson

DerGroncki, 2016-04-26:
May I ask you to describe the steps for creating the client and server certificates?

Thanks you.

Michael

robinson, 2016-04-27:
Hi Michael,
In order to generate certificate manually is complicated, therefore 
I am using the sample application from OPC Foundation, as following:
https://opcfoundation.org/developer-tools/developer-kits-unified-architecture/sample-applications
I suggest to install the app, then copy folders from: 
C:\ProgramData\OPC Foundation\CertificateStores\MachineDefault
into this folder:
C:\Users[user]\AppData\Local\Packages[packageName]\LocalState\OPC Foundation\CertificateStores\MachineDefault.

Hope that helps,

DerGroncki, 2016-04-27:
Hello Robinson,

Thanks for your answer. This was very helpful. Now both applications are running.

I tried to connect the “sample client” to the “sample server”. First I started the server. After the status of the server changed to “running” I started the client, selected the sample server from the pull down menu and clicked “connect”. 
In the popup-window the following message is displayed:
Attempting to read latest configuration option because the server is not available.
After a while it changed to:
Configuration options may not be correct because the server is not available.
It is not possible to connect to the server.

Have you tried to connect the sample server with the sample client?

Thanks for any hint.

Michael

robinson, 2016-04-29:
Hi Michael,

I have not seen this issue. Anyway, this is just workaround solution, it is not resolved the root cause.The generated keys functions which is  still there, see this:
https://github.com/OPCFoundation/UA-.UWP-Universal-Windows-Platform/blob/979764a24eddf82c9ec2f8d6785214dcc8ea2b93/Stack/Core/Security/Certificates/CertificateFactory.cs
It is calling   executablePath += "\\Bin\\Opc.Ua.CertificateGenerator.exe";
I believe it is just runnable with windows-based and inapplicable for ARM in Windows Core IOT. 
I think, there are 2 solutions:
1. using OpenSSL to generate keys.
2. rewrite the CertificateGenerattor tool by UWP.

Hope that helps,
Robinson

DerGroncki, 2016-04-29:
Hello Robinson, Thanks for your answer. I will try to get my head around openssl.

barnstee, 2016-05-13:
You don't need OpenSSL. Please use robinson's suggestion (that's what I use):
I am using the sample application from OPC Foundation, as following:
https://opcfoundation.org/developer-tools/developer-kits-unified-architecture/sample-applications
 I suggest to install the app, then copy folders from: 
 C:\ProgramData\OPC Foundation\CertificateStores\MachineDefault
 into this folder:
 C:\Users[user]\AppData\Local\Packages[packageName]\LocalState\OPC Foundation\CertificateStores\MachineDefault.