OPCFoundation/UA-.NETStandard #4: There is no cert with subject UA Sample Client in the configuration

DerGroncki, 2016-04-26:
I am getting the following error when I try to run the example projects:

(…) There is no cert with subject UA Sample Client in the configuration.
 Copy cert to this location: (...) \OPC Foundation\CertificateStores\MachineDefault

Any hint how to create the certificates for the client and the server would be great. Is it the same procedure as described here “Trusting Applications built with UA .NET API”.

Thanks.
Michael

K47S, 2016-04-27:
You should go to  https://opcfoundation.org/developer-tools/developer-kits-unified-architecture/sample-applications/ and download version 1.02.334.5 of the samples. Install it.
Included is a Opc.Ua.CertificateGenerator.exe which should create the certificate for you. 
The Stack automatically checks for the file on your pc (Check [here](https://github.com/OPCFoundation/UA-.UWP-Universal-Windows-Platform/blob/master/Stack/Core/Security/Certificates/CertificateFactory.cs#L362) for source). If it doesn't work you should search it and copy it to your project bin folder.

DerGroncki, 2016-04-27:
Thanks. I found the generator here:
C:\Program Files (x86)\Common Files\OPC Foundation\UA\v1.0\Bin
Now I will try your suggestion.

DerGroncki, 2016-04-28:
Ok, I found the only location for the generator "Opc.Ua.CertificateGenerator.exe" which is working seems to be the path
\AppData\Local\Packages\PackageName\LocalState\

A new certificate is only created if "dontCreateNewCertificate" is set to false:
\UA-.UWP-Universal-Windows-Platform\SampleApplications\SDK\Configuration\ApplicationInstance.cs
bool dontCreateNewCertificate = true; => must be set false

But I still get an error:
Server not started. Exit. Could not create a certificate via a proxy: Input file was not processed properly.

This is because the fist line of the tmp-file is "-cmd issue" which is created here:
\UA-.UWP-Universal-Windows-Platform\Stack\Core\Security\Certificates\CertificateFactory.cs

methode public static async Task<X509Certificate2Collection> CreateCertificateViaProxy

At the beginning a temp file is created and the first line written to the file is
writer.WriteLine("-cmd issue", storePath);

Some lines later the first line of the temp file is read and compared
if (result.StartsWith("-cmd"))
{
  throw new ServiceResultException("Input file was not processed properly.");
}

Am I totally wrong here? 

Any hint would be great.

Thanks.

Michael

DerGroncki, 2016-04-29:
Ok, the CertificateGenerator  is never invoked. This part of the program is missing or is not converted yet, respectively

Michael

closed 2016-04-29 by DerGroncki

barnstee, 2016-05-13:
CertificateGenerator.exe (or any exe for that matter) cannot be invoked from UWP apps. Therefore, the certificate must be generated by executing CertificateGenerator manually (via command prompt).

eoursel, 2016-05-21:
I still have an issue with that. I took the certificates generated by the .NET stack samples to copy them in the LocalState folder. Now the server and the client are starting without any error message. But when i want to connect to the server with an UA client, it says that serverCertificate is null. That is to say that the server certificate is not returned to the client, right? Anyway, there is something to do in order to enhance certificate handling by UWP apps.