OPCFoundation/UA-.NETStandard #9: Error starting server app with access denid

eoursel, 2016-05-14:
Looks like i can't start the server from Visual Studio debug mode (same error out of visual studio).

Looks like the socket can't be opened because of a missing permission 

`   at System.Net.Sockets.Socket.DoBind(EndPoint endPointSnapshot, SocketAddress socketAddress)
   at System.Net.Sockets.Socket.Bind(EndPoint localEP)
   at Opc.Ua.Bindings.UaTcpChannelListener.Start()`

The socketErrorCode is AccessDenied. DId i miss something in the configuration of the application? 

Running Windows 10 Insider Preview 10.0.14342 

Regards,

mregen, 2016-05-16:
@eoursel this error may indicate that another app already uses the listening port of the server. Depending on the configuration, if another OPC sample, e.g. the .Net one, is already running on the device, it may have already used the port.

eoursel, 2016-05-20:
My mistake. You are right This problem disappeared after a reboot.

closed 2016-05-20 by eoursel