OPCFoundation/UA-.NETStandard #11: SampleServer does not return endpoints

eoursel, 2016-05-22:
Just trying to connect the UWP SampleServer. Tested with uaexpert or another UA client (.NET) or node-opcua. Looks like the server does not return the endpoints. Also with the UA UWP client, thhe exception cannot fetch endpoint is triggered. 

uaexpert: 
Discovery GetEndpoints on opc.tcp://zion:51210/UA/SampleServer failed (BadTimeout)

mregen, 2016-06-10:
@eoursel Sorry for the late response. We need to repro the issue and we are working on it.

mregen, 2016-06-15:
@eoursel Did you try to connect from the same machine where the UWP Sample Server was running? UWP apps do not allow connections across localhost or from the same host name. This is part of the UWP security model. But you can connect from with any other OPC client across the network.

closed 2016-07-15 by barnstee