OPCFoundation/UA-.NETStandard #12: Roadmap to NanoServer/CoreCLR

cmello, 2016-05-24:
Hi!

Please I would like to know if there are plans to support Windows Nano Server. If I understand correctly Nano Server supports CoreCLR but not UWP.

As a bonus, CoreCLR support would allow to run the same code in Linux/Mac OS.

barnstee, 2016-05-31:
Hi cmello,
Good question: We are in fact working on a .Net platform standard OPC UA stack (which is platform independent) and we will refactor the UWP stack to use this .Net platform standard stack. We will also have implementations for CoreCLR and ASP.Net 5 (to allow OPC clients and servers to be run in the cloud).
More info here: [https://github.com/dotnet/corefx/blob/31e81793c9c21e2b5e165e0561b8169131f89f1f/Documentation/architecture/net-platform-standard.md]
This also enables OPC UA clients and servers to be implemented not just on UWP or ASP, but also on Mono and Xamarin. More info will be announced at OPC day Europe in Munich next month.
Cheers,
Erich

closed 2016-05-31 by barnstee

cmello, 2016-05-31:
Thank you very much @barnstee , we appreciate a lot your attenion. Best regards!