OPCFoundation/UA-.NETStandard #15: Pub/Sub Specification

zerox1212, 2016-06-08:
The ReadMe says  the publish example doesn't follow the OPC UA Pub/Sub specification. Where can I find the specification for Pub/Sub implementation on a protocol for AMQP?

K47S, 2016-06-09:
Hello,

as far as I know (and as the text above states) there isn't specification for pub/sub at the moment. I believe it in the making currently and takes some more time.

zerox1212, 2016-06-09:
That is what I figured. I am interested how it will be done. Having every subscribed node's data change create a AMQP message probably isn't a great solution. Services like Azure, which charge per message, will rapidly become very expensive.

Thanks for the response.

dk2129, 2016-06-10:
The OPC UA PubSub specification is still work in progress with a Release Candidate anticipated by end of 2017. At that time the specification will be public. The current drafts are accessible for OPC-UA working group members only.
As a general note: Like in any Publish-Subscribe model, the Publisher in OPC UA does not know the Subscribers. The publishing schedule and the items to be published are under control of the Publisher and there is no relationship to subscriptions created by an OPC UA Client. It addresses completely new use cases. You can find some basic information here: http://www.automation.com/automation-news/industry/opc-ua-is-enhanced-for-publish-subscribe-pubsub. There is also an open source project: https://github.com/mpostol/OPC-UA-OOI/blob/master/Networking/SemanticData/README.md.

zerox1212, 2016-06-10:
I'm looking forward to possibly implementing pub sub on another open source OPC UA project written in python. Thank you for the information.

barnstee, 2016-06-13:
Once the spec is officially available, we will of course update the sample to be spec complaint.

closed 2016-07-15 by barnstee