OPCFoundation/UA-.NETStandard #16: CerticateGenerator tool

jayantad, 2016-06-24:
Hi,

Please replace the below step:
Get from OPC Misc Tool the Certificate generator tool. (build the solution and get the Opc.Ua.CerticateGenerator.exe)

instead add the step:
Go to https://opcfoundation.org/developer-tools/developer-kits-unified-architecture/sample-applications/ and download and install version 1.02.336.0. The Opc.Ua.CertificateGenerator.exe will be installed on the machine.

Thanks

K47S, 2016-06-24:
I think it's up to what you prefer maybe they can link both possibilities. If you get the CerticateGenerator from the repo it's always the newest version. The version on the OPC Foundation page is almost a year old. The installer just makes it easier for you to install the CerticateGenerator.

jayantad, 2016-06-27:
I agree but the repo for CertficateGenerator is broken (ANSI C projects are missing) and it is not possible to build the exe. 
Also i think the CertificateGenerator has not changed much in last few years.

barnstee, 2016-06-27:
The ANSI C projects are coming, stay tuned. In the mean-time, please use the pre-built one from the OPC Foundation page.

dk2129, 2016-06-27:
The Certificate Generator is an independent tool. The OPC Foundation has published this tool in a separate repository. See https://github.com/OPCFoundation/Misc-Tools.

closed 2016-07-15 by barnstee

NicoKno, 2019-06-13:
The above link https://opcfoundation.org/developer-tools/developer-kits-unified-architecture/sample-applications/ is dead.

AlinMoldovean, 2019-06-21:
@NicoKno ,

Try this:
https://opcfoundation.org/developer-tools/samples-and-tools-unified-architecture