OPCFoundation/UA-.NETStandard #19: Decoupling functionality.

mpostol, 2016-07-15:
It is especially important for Core library. As far as today by design the library is collection of tightly coupled functionalities. Partially it is used in other application, e.g. [UA-ModelCompiler](https://github.com/OPCFoundation/UA-ModelCompiler), so it also needs decoupling.

To follow the separation of concern principle it seems that the following functionality shall be decoupled:
 - common definitions 
 - stack (protocols)
 - services
 - address space management
 - tracing
 - configuration
 - TBD 

For some of them (e.g. protocols, tracing, configuration) Dependency Injection technique shall be considered.

barnstee, 2016-07-15:
Agreed. Also, all the certificate management (which is spread quite far right now) can be consolidated under a "CryptoProvider" interface. This will make implementing new cert stores easier.

mpostol, 2016-07-20:
Pub/Sub (if implemented) must be also loosely coupled. My point is that it should be required to support possibility that the publisher distributes data not only coming from the OPC UA Address Space. In other word – the databinding functionality should be flexible enough to gather data from any generic data source (thing referring to the IoT terminology). More about the architecture and usage scenarios can be found at:

[Networking of SemanticData Library]( https://github.com/mpostol/OPC-UA-OOI/tree/master/Networking/SemanticData#networking-of-semanticdata-library)

barnstee, 2016-09-06:
Right now, Pub/Sub is implemented entirely in the OPC Publisher samples, including AMQP transport. The only component required from the Core or SDK library is the JSON parser. This means it is alreasy loosly coupled.

mpostol, 2016-09-06:
Not sure, in [Publisher project](https://github.com/OPCFoundation/UA-.NETStandardLibrary/tree/master/SampleApplications/Samples/Publisher) we can find:
`
  <ProjectReference Include="..\ClientControls\Opc.Ua.Client.Controls.csproj">
      <Project>{0bacfd1f-8bd1-45e2-81ad-f9dbb5fbe028}</Project>
      <Name>Opc.Ua.Client.Controls</Name>
    </ProjectReference>
    <ProjectReference Include="..\Controls\Opc.Ua.Sample.Controls.csproj">
      <Project>{5192cb9a-d4fc-41ef-839f-d9e08c17f221}</Project>
      <Name>Opc.Ua.Sample.Controls</Name>
    </ProjectReference>
  </ItemGroup>
  <ItemGroup>
    <Reference Include="Opc.Ua.Client, Version=1.2.334.6, Culture=neutral, processorArchitecture=MSIL">
      <SpecificVersion>False</SpecificVersion>
      <HintPath>..\..\SDK\Opc.Ua.Client\bin\Debug\uap10.0\Opc.Ua.Client.dll</HintPath>
    </Reference>
    <Reference Include="Opc.Ua.Configuration, Version=1.2.334.6, Culture=neutral, processorArchitecture=MSIL">
      <SpecificVersion>False</SpecificVersion>
      <HintPath>..\..\SDK\Opc.Ua.Configuration\bin\Debug\uap10.0\Opc.Ua.Configuration.dll</HintPath>
    </Reference>
    <Reference Include="Opc.Ua.Core, Version=1.2.335.0, Culture=neutral, processorArchitecture=MSIL">
      <SpecificVersion>False</SpecificVersion>
      <HintPath>..\..\..\Stack\Opc.Ua.Core\bin\Debug\uap10.0\Opc.Ua.Core.dll</HintPath>
    </Reference>
    <Reference Include="Opc.Ua.Sample, Version=1.0.0.0, Culture=neutral, processorArchitecture=MSIL">
      <SpecificVersion>False</SpecificVersion>
      <HintPath>..\Opc.Ua.Sample\bin\Debug\uap10.0\Opc.Ua.Sample.dll</HintPath>
    </Reference>
    <Reference Include="Opc.Ua.Server, Version=1.2.334.6, Culture=neutral, processorArchitecture=MSIL">
      <SpecificVersion>False</SpecificVersion>
      <HintPath>..\..\SDK\Opc.Ua.Server\bin\Debug\uap10.0\Opc.Ua.Server.dll</HintPath>
    </Reference>
`
In this context, loosely coupled mens that we can use late binding. Hard-coded references to the whole OPC UA staff means that any changed needs rebuilding of the whole stack. 

The firs step before decoupling should be using Nuget packaes. 

By the way I cannot find subscriber implementation. 
Sorry for formatting of the XML snipped.

barnstee, 2016-09-06:
NuGet packages are published but hidden so far, until the OPC Foundation has a chance to create landing pages for the licencing terms. Shouldn't take much longer.

barnstee, 2016-09-06:
Also, for our OPC Publisher sample, Azure IoT Hub is the subscriber and the data can be visualized using the telemetry web app sample which can be found in the root directory.

mpostol, 2016-09-06:
@barnstee to be sure, is the Azure IoT Hub the:

* AMQP subscriber or 
* Pub/Sub over AMQP subscriber. 

In other words is the the Azure IoT Hub the Pu/Sub protocol aware? What parts of the specification has been used to implement the dedicated functionality of Pub/Sub in the Azure IoT Hub.

It is relevant for me because decoupling the Pub/Sub over AMQP protocol suit will allow to bind it and reuse it in my project where publisher and subscriber roles are symetric.

barnstee, 2016-09-07:
Azure IoT Hub is just the AMQP subscriber and doesn't look at the Pub/Sub payload at all. The telemetry website then reads the messages from the IoT Hub and parses them (i.e. it is Pub/Sub aware).

mpostol, 2016-09-07:
From what you said I understand that talking about interoperability in context of Azure IoT Hub and telemetry website only the following standards apply:

* AMQP - communication - data transfer
* JSON - payload - data access

Are us sure that the Azure IoT Hub decoder supports all OPC UA build-in tupes, e.g. Uuid/GUID, DataTime, NodeId, etc. For example look at this snipped from [public class JsonEncoder : IEncoder, IDisposable](https://raw.githubusercontent.com/OPCFoundation/UA-.NETStandardLibrary/c02d1ca8b4cf1c29e3c631442b8ba9955d52c960/Stack/Opc.Ua.Core/Types/Encoders/JsonEncoder.cs):

~~~~
/// <summary>
/// Writes a GUID to the stream.
/// </summary>
public void WriteGuid(string fieldName, Uuid value)
{
    if (value == Uuid.Empty)
    {
        WriteSimpleField(fieldName, null, false);
    }
    else
    {
        WriteSimpleField(fieldName, value.ToString(), true);
    }
}
/// <summary>
/// Writes a GUID to the stream.
/// </summary>
public void WriteGuid(string fieldName, Guid value)
{
    if (value == Guid.Empty)
    {
        WriteSimpleField(fieldName, null, true);
    }
    else
    {
        WriteSimpleField(fieldName, value.ToString(), true);
    }
}
~~~~

The pub/sub spec proposes extension of the JSON specification for encoding of the not supported OPC UA Build-in types implemented by overriding `ToString()`.

My concern is if there is any part of the pub/sub spec promoting interoperability between OPC UA and Azure IoT Hub and telemetry website .

Concluding, can we say:
> pub/sub specification is **not relevant** in context of OPC UA and Azure IoT Hub interoperability implementation and conformance testing.

It is very important question because pub/sub is recognized as the **IoT/IIoT enabler** by the community.

barnstee, 2016-09-07:
Correct, in context of Azure IoT Hub interoperability implementation, the Pub/Sub spec does not need to be considered (as IoT Hub works on the AMQP transport layer only). It only needs to be considered when looking at the JSON payload (which is what the telemetry website does). Right now, the telemetry website only uses the members from the OPC UA MonitoredItem class. As folks implement other Azure web sites or services depdentent on Pub/Sub data, this will most likely need to be extended.

mpostol, 2016-09-09:
Because the pub/sub is not yet matured let’s go back to the main topic of this thread, namely the core library architecture.

I have investigated the certificate management in details and have two remarks:
- it is dependent on the configuration stuff and the configuration stuff provides helpers related to certificate management (looks like circular dependency)
-  this functionalities depends on the environment class  and user interface so unit testing is very difficult.

It seems to me that before putting apart logical units a master plan must be designed. Additional problem arises how to synchronize this library with the root project UA-.NET.

barnstee, 2016-12-09:
We decided NOT to re-architect this stack to keep it the same as the old .Net stack and make the transition to the new stack easier as well as being able to merge changes between the two stacks easily.

closed 2016-12-09 by barnstee