OPCFoundation/UA-.NETStandard #20: Unit Test to improve quality

mpostol, 2016-07-15:
After decoupling of selected functionality UT shall be added at least to Core library/libraries.

barnstee, 2016-07-15:
Right on. Christian and Martin are on the hook for this.

mregen, 2016-10-05:
We started testing the reference server against CTT.

mpostol, 2016-10-05:
Great, but CTT testing and Unit Testing are not comparable and replaceable- different approach, different goals, different scope, etc.. To avoid confusion for CTT we need independent issue/task. CTT shall be applied just before release to prove interoperability against specification, but not to prove internal code correctness.

barnstee, 2016-10-05:
Christian from Softing is currently integrating Unit Tests.

barnstee, 2016-12-09:
Since the unit tests are Softing's, we can't check them in here, but they do exist!

closed 2016-12-09 by barnstee