OPCFoundation/UA-.NETStandard #21: test results on compatible net core OS.

eoursel, 2016-07-16:
Looks like you made a huge progress in the last commit. I made some tests on Ubuntu 2 weeks ago with the Marton's repository and I remember there were some trouble on this platform. It seems you fixed everything but I didn't test because I am on vacation. 

It would be nice to have in the wiki something like tests results on the different OS supported by the .net core platform. Tests should include unit tests and OPC Foundation CTT test cases.

eoursel, 2016-07-16:
Martin not Marton, sorry. Using github on my smartphone. About unit tests, the best example is what was written for node-opcua. The problem is that unit tests are supposed to be written before the source code itself. Write unit tests after the source code will be a challenging task.

Regards

Eric

barnstee, 2016-07-17:
Martin is testing on Ubuntu (I think!) right now, actually. We had it working before, so this is a regression test. Agreed that unit tests should ideally come before the code (test first!), but in this case we were porting from the old .Net stack. We are currently porting the unit tests from the old .Net stack.

barnstee, 2016-07-20:
Result: There was an config issue with Linux (and some "\" which needed to be "/", but it is working now.

closed 2016-07-20 by barnstee