OPCFoundation/UA-.NETStandard #24: DNS fails on Linux

mregen, 2016-07-19:
on MacOS (and likely other Linux flavors) DNS resolution fails.

> Project NetCoreConsoleClient (.NETCoreApp,Version=v1.0) was previously compiled. Skipping compilation.
> .Net Core OPC UA Console Client sample
> 1 - Create an Application Configuration.
>     WARN: missing application certificate, using insecure connection.
> 2 - Discover endpoints of OPC UA server.
> Could not fetch endpoints from url: opc.tcp://opc-server:51210/UA/SampleServer
> Reason = One or more errors occurred. (This platform does not support connecting sockets to DNS endpoints via the instance Connect and ConnectAsync methods, due to the potential for a host name to map to multiple IP addresses and sockets becoming invalid for use after a failed connect attempt. Use the static ConnectAsync method, or provide to the instance methods the specific IPAddress desired.)
> Exit due to Exception: One or more errors occurred. (One or more errors occurred. (This platform does not support connecting sockets to DNS endpoints via the instance Connect and ConnectAsync methods, due to the potential for a host name to map to multiple IP addresses and sockets becoming invalid for use after a failed connect attempt. Use the static ConnectAsync method, or provide to the instance methods the specific IPAddress desired.))

mregen, 2016-07-23:
Testing fix here: 5733d9c77bb8af4dcbd9ccf36161229b39c33250

Suman3484, 2016-08-02:
Hi,

I am getting the below error 

"This platform does not support connecting sockets to DNS endpoints via the instance Connect and ConnectAsync methods, due to the potential for a host name to map to multiple IP addresses and sockets becoming invalid for use after a failed connect attempt. Use the static ConnectAsync method, or provide to the instance methods the specific IPAddress desired"

The code runs fine in windows, but when I tried to run in linux centos, I am getting this error.

Any help is much appreciated here.

Here is my TCP IP Connect Method .

   public ASTCPIPStream(ServerEntry objEntry)
        {

            try
            {
                Console.WriteLine("Entering ASTCPIPStream " );

                TcpClient client = new TcpClient();
                client.Client.Connect(objEntry.m_serverName, objEntry.m_nPort); //make connection
                //client.Client.SetSocketOption(SocketOptionLevel.Socket, SocketOptionName.KeepAlive, true);
                if (objEntry.m_bIsSecure && objEntry.m_strCertificationPath.Trim().Length > 0) //SSL connection !!
                {

                    SslStream objStream = new SslStream(client.GetStream(), false,
                        new RemoteCertificateValidationCallback(ASCSServerCertificateValidator),
                        new LocalCertificateSelectionCallback(ASCSCertificateSellector));

                    X509Certificate objCert = new X509Certificate(objEntry.m_strCertificationPath);
                    X509CertificateCollection objCertCollection = new X509CertificateCollection();
                    objCertCollection.Add(objCert);
                    objStream.AuthenticateAsClientAsync(objEntry.m_strSSLServerName, objCertCollection, objEntry.m_bIsTLS ? SslProtocols.Tls12 : SslProtocols.Ssl3, false);

                    System.IO.BufferedStream obj = new BufferedStream(objStream);
                    SetStream(ref obj);
                }
                else
                {
                    //System.IO.Stream obj = (System.IO.Stream)client.GetStream();
                    System.IO.BufferedStream obj = new BufferedStream(client.GetStream());
                    SetStream(ref obj);
                }
            }

            catch (Exception ex)
            {
                Console.WriteLine("Error in ASTCPIPStream" + " - Message : " + ex.Message + "StackTrack : " + ex.StackTrace);
            }
        }

mregen, 2016-08-04:
@suman3484 Please make sure you have fix name resolution on Linux #33. If there is still an issue please let me know...

barnstee, 2016-08-05:
Issue appears to have been fixed as I did not hear back from @Suman3484

PR #33 (2016-07-25): fix #24 name resolution on Linux
Fix #24 DnsEndpoint caused an error on dotnet Linux runtimes. Change the connect strategy to use GetHostAddressesAsync, then try the first Ipv6 and Ipv4 address in the list to connect. Tested on Console, Windows, UWP and Linux samples/runtimes.
commit 8cd9eedc (2016-07-25): Merge pull request #33 from mregen/linux_connect

fix #24 name resolution on Linux
closed 2016-08-05 by barnstee