OPCFoundation/UA-.NETStandard #44: Add support for a TPM cert store

barnstee, 2016-08-05:
Since the Windows cert store had to go (not cross platform), we Need a secure replacement. TPM is the obvious choice, but we Need a cross platform, .Net Standard-compatible TPM library for that. Looking at TSS right now (https://www.microsoft.com/en-us/download/details.aspx?id=52507).

barnstee, 2016-09-19:
First step complete: Port TSS to .Net Standard: https://github.com/Microsoft/TSS.MSR/pull/5

BoeseB, 2016-10-18:
Are there any alternatives to the hardware solution with a TPM cert store?

barnstee, 2016-10-18:
Yes, you could use a firmware TPM (instead of a discrete TPM), if your hardware platform supports it, or you could keep the private keys in the file system but encrypt the file system, if your OS supports it. Some OSes also privide a secure store in software (but hardware is more secure, of course).

barnstee, 2016-12-09:
Progress is slow, but it's coming along: https://github.com/barnstee/UA-.NETStandardLibrary/blob/master/Stack/Opc.Ua.Core/Security/Certificates/TPMCertificateStore.cs
There simply aren't enough hours in the day...

barnstee, 2017-02-16:
At least we now have cross-platform OS-cert-store support back in, in form of the X509CertificateStore I just added. This works for both Linux and Windows and works with private keys, too, so we have an OS-supported secure storage area. TPM would be yet another level of security, still waiting on Linux support in https://github.com/Microsoft/TSS.MSR. It is now .Net Standard compatible, but so far just supports Windows...

BoeseB, 2017-02-17:
I just reviewed your Commit and was asking myself why the `X509CertificateStore ` class needs an Open and Close method. Wouldn't it be better if the store get's opened when the instance is created and automatically Closed on Disposing? This way i wouldn't have to watchout for call order. What is the purpose of Open and Close? If it's critical that it's only open for one action you could just put the store instance into a using block.

barnstee, 2017-02-17:
You are right, the open and close is actually not needed for the X509Store. It is part of the ICertificateStore interface, though, so I needed to implement it. It is however already closed on disposing...

BoeseB, 2017-02-19:
You could remove those two methods in the interface, like you already did with other members. What store would require the explicit open and close?
On each instantiation of the store the Open method follows immediately in your submitted changes.

barnstee, 2017-02-20:
All the other store implementations. It is part of the interface to the stores. I don't want to change this now.

eoursel, 2019-12-16:
Hi all 
What is the status of the TPM support in the Net Core implementation? Is it integrated in the source code? What i need is a storage of the private keys in the TPM Hardware only on Windows. If I plan to use the Windows Certificate Store only is it working ?

barnstee, 2019-12-16:
Yes, Windows-only is working and it is in my branch. I tested it on Windows with a Infinion TPM. It is not integrated into master yet, however, but feel free to create a pull request.

mregen, 2021-01-12:
Outdated...

closed 2021-01-12 by mregen