OPCFoundation/UA-.NETStandard #45: Trace file not created

eoursel, 2016-08-14:
On Ubuntu 16.04 looks like the path in Opc.Ua.SampleServer.Config.xml is not used and the trace in created in /tmp/trace.log.  

Testing with 1.0.0-preview2-003121

barnstee, 2016-08-17:
@eoursel:
<OutputFilePath>Logs\Opc.Ua.SampleServer.log.txt</OutputFilePath> should probably be <OutputFilePath>Logs/Opc.Ua.SampleServer.log.txt</OutputFilePath>
Can you try this out?

eoursel, 2016-08-17:
Yes. I tried. But it does not change anything. The trace file is always created in /tmp/trace.log

barnstee, 2016-08-18:
OK, thanks. We'll have a look...

eoursel, 2016-08-18:
After a comparison with the original .NET code, i found that the method Utils.SetTraceLog() is not called by the stack in TraceConfiguration class. So the trace system is not correctly initialized.

eoursel, 2016-08-19:
@barnstee, I found some issues in Opc.Ua.Core/Types/Utils/Utils.cs. While i could fix that easily, i think the code could be enhanced and secured with the usage of a portable framework which is natively supported on the .NET core platform and will be extensible such as the Microsoft.Extensions.Logging package (https://msdn.microsoft.com/en-us/magazine/mt694089.aspx). 

Let me know what you think about that.

barnstee, 2016-08-19:
@eoursel, thanks for figuring this one out. Let's do this: Pease submit a pull request with the easy fix and then take a stab at enhancing the logging with the new Microsoft.Extensions.Logging package. When making a major change like this, it is always important to make sure the packages are supported and working properly on all target frameworks, i.e. .Net Core, UWP, .Net 4.6.

barnstee, 2016-08-19:
I believe I fixed the trace log issue. Please pull. There were several small fixes to be made. The difficulty is that in UWP, you don't have write access to the application directory, so you need to write the trace file to temp. If you want the tracefile somplease else, you need to specify an absolute path.

closed 2016-09-05 by barnstee