OPCFoundation/UA-.NETStandard #46: Problem with certificates generated by createcert.sh (openssl)

eoursel, 2016-08-14:
There is an issue with the certificated generated with createcert.sh on Ubuntu 16.04 while the certifcate generated by OPC CertificateGenerator (generated on windows and copied to the linux host). Find below the trace file.  

```
10:07:35.940 Loading application configuration file. /home/eric/lab/opc_f/UA-.NE
TStandardLibrary/SampleApplications/Samples/NetCoreConsoleServer/Opc.Ua.SampleSe
rver.Config.xml
10:07:36.377 Could not load private key for certificate UA Sample Server OpenSsl
CryptographicException 'error:23076071:PKCS12 routines:PKCS12_parse:mac verify f
ailure'
10:07:36.387 Could not load private key certificate from file: arizona.pfx OpenS
slCryptographicException 'error:23076071:PKCS12 routines:PKCS12_parse:mac verify
 failure'
10:07:36.396 Checking application instance certificate.
10:07:36.397 Could not load private key certificate from file: arizona.pfx OpenS
slCryptographicException 'error:23076071:PKCS12 routines:PKCS12_parse:mac verify
 failure'
10:07:36.400 Could not load private key certificate from file: arizona.pfx OpenS
slCryptographicException 'error:23076071:PKCS12 routines:PKCS12_parse:mac verify
 failure'
10:07:36.403 ServiceResultException:One or more errors occurred. (Cannot access 
certificate private key. Subject=CN=UA Sample Server, DC=arizona)

```

barnstee, 2016-08-17:
@eoursel. Martin, who wrote the script, is currently on vacation, but will have a look when he returns in 2 weeks' time.

mregen, 2016-08-30:
@eoursel sorry I can't repro this here on Ubuntu 16.04. Please check this:
i) there should not be a password set on the private key. Just hit twice return.
ii) rename or delete the ` ./OPC Foundation/ ` folder before `./createcert.sh` is run to make sure there are only the new certs in the CertificateStore.
If it still doesn't work could you send me the failing .der & .pfx certs?

eoursel, 2016-08-31:
@mregen you are right that's the password of the private key. I would recommend to add the option -passout like 

`openssl pkcs12 -export -in cert.pem -inkey cert.key -out $MYHOSTNAME.pfx -nodes -passout pass: `

in order to avoid entering a password. As a matter of fact, i would try to automate the creation of a docker container with the UA server and it's a mandatory requirement to have a completely automated with no operator interaction. 

Anyway, there is at least a cyber security issue if we can't use a pgarrivate key protected by a password to start a server.  Ok i know that the TPM library @barnstee mentioned is certainly a clever solution. 

Regards 

Eric

mregen, 2016-09-01:
@eoursel Thanks, good point, I will fix the scripts. The scripts are only meant to be used to run the samples, not in production. 
For the pw support for the private keys we should have a feature request to make sure it is properly supported if required.

mregen, 2016-09-01:
#51 to improve cert handling.

closed 2016-09-05 by barnstee