OPCFoundation/UA-.NETStandard #47: BufferManager TakeBuffer method expecting more than the maximum configured buffer size

solidrevolution, 2016-08-17:
Now that there is an OPC UA implementation targeting the Standard library we wanted to write a mobile client implementation using the Xamarin platform. 

When trying to connect a OPC UA server we get the following exception

> Specified argument was out of the range of valid values.
---   at System.ServiceModel.Channels.BufferManager+DefaultBufferManager.TakeBuffer (Int32 bufferSize) [0x0001f] in /Users/builder/data/lanes/3540/1cf254db/source/mono/mcs/class/System.ServiceModel/System.ServiceModel.Channels/BufferManager.cs:174 
---   at Opc.Ua.Bindings.BufferManager.TakeBuffer (Int32 size, System.String owner) [0x0002c] in XXX\Repos\UA-.NETStandardLibrary\Stack\Opc.Ua.Core\Stack\Bindings\BufferManager.cs:162 
---   at Opc.Ua.Bindings.TcpMessageSocket.ReadNextMessage () [0x00022] in XXX\Repos\UA-.NETStandardLibrary\Stack\Opc.Ua.Core\Stack\Tcp\TcpMessageSocket.cs:348 
---   at Opc.Ua.Bindings.TcpClientChannel.OnConnectComplete (System.Object sender, System.Net.Sockets.SocketAsyncEventArgs e) [0x00076] in XXX\Repos\UA-.NETStandardLibrary\Stack\Opc.Ua.Core\Stack\Tcp\TcpClientChannel.cs:747 

The exception is thrown from the `TakeBuffer` method of the `DefaultBufferManager` class of the Mono framework. 
```
public override byte [] TakeBuffer (int bufferSize)
{
	if (bufferSize < 0 || (max_size >= 0 && bufferSize > max_size))
		throw new ArgumentOutOfRangeException ();

	int l2 = log2 ((uint) bufferSize);
	if (l2 > log_min)
		l2 -= log_min;

	List<byte []> returned = buffers [l2];
	if (returned == null || returned.Count == 0)
		return new byte [bufferSize];
	
	foreach (var e in returned){
		if (e.Length >= bufferSize){
			hits++;
			returned.Remove (e);
			return e;
		}
	}
	return new byte [bufferSize];
}
```
After some further analysis we found that `BufferManager` implementations of the [Mono framework](https://github.com/mono/mono/blob/ef407901f8fdd9ed8c377dbec8123b5afb932ebb/mcs/class/System.ServiceModel/System.ServiceModel.Channels/BufferManager.cs) and the [.NET framework](https://github.com/mono/mono/blob/d70777a3332af2d630d24adf620c2e548b92b56a/mcs/class/referencesource/System.ServiceModel/System/ServiceModel/Channels/BufferManager.cs) differ. While Mono implementation tests against the configured `max_size` value and throws an exception there is no such control in the .NET framework code. 

    public byte[] TakeBuffer(int size, string owner)
    {
        if (size > Int32.MaxValue - 5)
        {
            throw new ArgumentOutOfRangeException("size");
        }

        lock (m_lock)
        {
            #if TRACK_MEMORY
            byte[] buffer = m_manager.TakeBuffer(size+5);                             

            byte[] bytes = BitConverter.GetBytes(++m_id);
            Array.Copy(bytes, 0, buffer, buffer.Length-5, bytes.Length);                
            buffer[buffer.Length-1] = 0;

            m_allocated += buffer.Length;

            Allocation allocation = new Allocation();
            
            allocation.Id = m_id;
            allocation.Buffer = buffer;
            allocation.Timestamp = DateTime.UtcNow;
            allocation.Owner = owner;

            m_allocations[m_id] = allocation;
            
            return buffer;
            #else
            byte[] buffer = m_manager.TakeBuffer(size+1);               
            buffer[buffer.Length-1] = 0;
            return buffer;
            #endif
        }
    }
    
So the obvious question is how are we calling `TakeBuffer` with a buffer size bigger than the configured maximum size. The answer to this question lies in the [BufferManager](https://github.com/OPCFoundation/UA-.NETStandardLibrary/blob/master/Stack/Opc.Ua.Core/Stack/Bindings/BufferManager.cs) class of the stack. This class internally uses the built in `BufferManager` implementations of the underlying framework but on `TakeBuffer` method adds some more to the expected buffer size.

One of the many places this method is called is inside the `ReadNextMessage` method of the [TcpMessageSocket](https://github.com/OPCFoundation/UA-.NETStandardLibrary/blob/master/Stack/Opc.Ua.Core/Stack/Tcp/TcpMessageSocket.cs) class. This class sends the internal `m_receiveBufferSize` field to the `TakeBuffer` method. But examining where this field's value comes leads us to the [TcpClientChannel](https://github.com/OPCFoundation/UA-.NETStandardLibrary/blob/master/Stack/Opc.Ua.Core/Stack/Tcp/TcpClientChannel.cs) class.

    Socket = new TcpMessageSocket(this, BufferManager, Quotas.MaxBufferSize);
  
  As can be seen the value sent to the `TakeBuffer` is already the maximum buffer size and the `BufferManager` class is adding some more and expecting more than the maximum configured buffer size.

Are we missing something here? 

Thanks in advance

barnstee, 2016-08-17:
No, you're not. I came across this code as well and thought this was a bug. The overridden BufferManager implementation is only used for debugging purposes and to track memory leakes (which does work, if you're lucky enough that the memory leak is within code that uses it!). So I was thinking of throwing out this implementation a while ago because, as you have discovered, the implementation is platform-dependent and therefore unpredictable. Do you want to take a stab at either removing the code or fixing it and send a pull request?
A simple fix would be:
System.ServiceModel.Channels.BufferManager.CreateBufferManager(maxPoolSize, maxBufferSize+1); and:
System.ServiceModel.Channels.BufferManager.CreateBufferManager(maxPoolSize, maxBufferSize+5);, if TRACK_MEMORY is defined.
Also, do you want to contribute a Xamarin sample (client or server)? This would complete our samples collection nicely...

solidrevolution, 2016-08-18:
I sent a pull request with the simple fix. But it seems like you mentioned `BufferManager `is causing some more problems :) We are getting exceptions from the `UnlockBuffer `method

If i understand you correctly you are saying `BufferManager `can be removed. I do not feel confident about changing that much but it seems i will have to give it a shot. Do you have anything in mind? Any sugestions?

Regarding the Xamarin sample we would love to contribute but i can not give an estimate since this is somewhat a side project for us atm. Do you have any requirements guidelines for that? How should it look like? which features should it have? etc

barnstee, 2016-08-18:
Thanks for the pull request, it's merged. What other exceptions are you experiencing?
Regarding the demo features, since this is a mobile client app, connecting (securely) to an OPC UA server, browsing its nodes and then creating a subscription on one or multiple nodes would be awesome. :-)

barnstee, 2016-08-18:
I could repro your exception and it can be ignored. It happens when a previous exception causes a double unlock on a buffer. Since the buffer lock is only used to see if a buffer is still in use when it is cleaned up, double locking or unlocking should not be an exception. I changed it to a trace warning instead, just now, please pull.

solidrevolution, 2016-08-22:
Great! 

We managed to create a session and read some values. 

Thanks!

closed 2016-08-22 by solidrevolution

brunoleos, 2016-09-30:
I'm having the argument exception issue also. I see you updated the repository code, but I need to stay with 1.02.334.5 release because (I think) it is the last that supports .NET 3.5. I couldn't compile the newer releases with .NET 3.5.

I'm going to try the fix you suggested in my old revision.

By the way, I'm using Unity, which uses a Mono compiler stick to .NET 3.5. I just need a simple .NET (framework 3.5 compatible) OPC client to communicate to a OpenModelica generated OPC server. I'm having a hard time trying to build the client code, as I don't need security, and I'm having issues with your library and the Mono compiler. I would be grateful If anyone could help with a simple solution.

brunoleos, 2016-09-30:
I had to change the fix to System.ServiceModel.Channels.BufferManager.CreateBufferManager(maxPoolSize, maxBufferSize+2);

My client is based in [http://stackoverflow.com/questions/30573689/opc-ua-minimal-code-that-browses-the-root-node-of-a-server/30625358#30625358](url)