OPCFoundation/UA-.NETStandard #53: Cannot find "Publish" button to start publishing the node's Pub/Sub encoded data to your IoTHub

lianxiwang, 2016-09-05:
Hi,

I followed README and everthing goes well, but got trouble of below step, "•Press the "Publish" button and the application will start publishing the node's Pub/Sub encoded data to your IoTHub". Because i cannot find any "Publish" button.

Appreciate your help here, thanks in advance.

mregen, 2016-09-05:
@lianxiwang Hi, which sample did you test? I think our readme might be out of date. e.g. in the Publisher.Net4 you start a subscription and its automatically publishing. Right click on the subscription pops up an enable/disable flag for the publisher.
On UWP the 'disconnect' button becomes 'publish' after selecting a node.

lianxiwang, 2016-09-06:
Thanks for your update and it is helpful for me.

closed 2016-09-06 by lianxiwang