OPCFoundation/UA-.NETStandard #54: OPC UA Android Client

harpreet-singh-bola, 2016-09-06:
Dear friends,

I would like to know if it is possible to create Android clients that connect to OPC UA server.

I tried to port the Client sample to my Android project but the references in the project points to libraries like Windows forms that are not needed/available for Android.

Another problem is that the connection to OPC UA server is made using some xml configuration files not exactly clear, is there a way to do it via code as in Android it would be more easy to do that way?

I searched through the internet but not found any Wiki or docs related to UA-.NETStandardLibrary, can any one please help?

mregen, 2016-09-06:
@harpreet-singh-bola Hi, please try the NetCoreConsoleClient sample first. It uses a minimal footprint to get connect to a OPC UA server without configuration files. The Mono project now also supports .Net Standard Library -->https://blog.xamarin.com/net-standard-library-support-for-xamarin/.
This is all brand new so we haven't been able to test with Xamarin yet, so please report any issues you may find. Thanks, Martin

barnstee, 2016-09-06:
@solidrevolution is also working on an Android/Xamarin client and he promised to publish a Xamarin sample client here on GitHub. Perhaps collaborate with him?

harpreet-singh-bola, 2016-09-06:
@mregen Thank you for the suggestions, I will try and let you know! Thanks again :)

@barnstee I would like to collaborate with him, how can I proceed? Can I contact him directly?

HSB

barnstee, 2016-09-06:
Sure.

harpreet-singh-bola, 2016-09-06:
How can I contact him? There is no e-mail id exposed on his profile. I will wait until he reads this issue.

barnstee, 2016-09-06:
He's on Twitter: https://twitter.com/solidrevolution

solidrevolution, 2016-09-06:
Hey ^^

Yes we are working on a Xamarin client and managed to deploy to an Android device and connect to some devices. For ios we managed to make it work on simulator but we are having some issues on deploying to devices.

On the contribution part we are having some issues with company policies i am not sure whether we get permission but we will see. In the mean time i can help you if you(@harpreet-singh-bola) plan on starting an implementation. Give me an email address and i will send you a summary of what we did.

Best regards

barnstee, 2016-09-06:
Sweet. Please also include me (erichb@microsoft.com).

harpreet-singh-bola, 2016-09-06:
Hello @solidrevolution ! Yes, I would love to have some point outs on how can I implement some basic functionalities first.

I am really thankful to you, my e-mail is: harpreetsingh.bola@gmail.com.

Regards,
HSB

CarlCrol, 2016-10-24:
@solidrevolution  hi would also like to get information on how you were able to connect to an OPC UA Server througj Xamarin.android. 

carl_abarintos@yahoo.com

pedosoul, 2016-11-16:
Hi I would like to get information too to use OPC UA client on Xamarin.Android

pedosoul@gmail.com

danielearrighi, 2017-02-20:
Hi I would like to have more infos to create a OPC UA client on Xamarin.Android. 
Something changed?
[daniele.arrighi@gmail.com](daniele.arrighi@gmail.com)
Thanks

solidrevolution, 2017-02-27:
I tried to summarize the steps we took to make the Xamarin OPC UA client work. I hope it helps. 

**Convert default template to work as a standard library project**

As of today even though the Xamarin platform supports .NET Standard library there is no project template built into the visual studio extension. With a few modifications the cross platform PCL template can be converted to target .NET Standard library. Below i am giving some references that explains the conversion:

https://blog.xamarin.com/net-standard-library-support-for-xamarin/
https://oren.codes/2016/07/09/using-xamarin-forms-with-net-standard/
https://xamarinhelp.com/dot-net-standard-pcl-xamarin-forms/
https://xamarinhelp.com/net-standard-xamarin-forms-gotchas/

But luckily there is no need to do the conversion manually an already converted and running hello world can be found in this repository - https://github.com/onovotny/SampleXamFormsWithNetStandard Just satisfy the requirements below and it should work.

    Visual Studio 2015 with Update 3
    .NET Core 1.0 for Visual Studio
    The latest Xamarin in the stable channel (configured correctly with the Android sdk and stuff)

**Add the necessary Opc.Ua dlls to the shared(portable) project**

I think the standard library opc ua dlls are not available from nuget instead we downloaded the stack from github and built our own dlls. Since we are after a client application we need the following two dlls

Opc.Ua.Client.dll
Opc.Ua.Core.dll

They can be found under the following folders

UA-.NETStandardLibrary\Stack\Opc.Ua.Core\bin\<Debug>\netstandard1.3
UA-.NETStandardLibrary\SampleApplications\SDK\Opc.Ua.Client\bin\<Debug>\netstandard1.3

With normal Xamarin, one needs to add same references to all the project types but with standard this is not the case. Our current application references the opc dlls only on shared(portable) project

**Satisfy dependencies for Opc.Ua**

Opc.Ua itself depends on some other libraries and since we are not using nuget we should satisfy them manually. We added the following dependencies using nuget to the shared(portable) project

Microsoft.AspNetCore.Server.Kestrel
Microsoft.AspNetCore.Server.Kestrel.Https
Newtonsoft.Json

**Write the code**

At this point the application should be buildable and deployable without problems. Now comes the part where the actual opc ua code is written. We used the NetCoreConsoleClient and the samples from the non standard library(https://github.com/OPCFoundation/UA-.NET) implementation to come up with our code.

From this point onwards it is lots of debugging/trying to overcome network/firewall issues. We used https://www.kepware.com/products/kepserverex/ as a opc ua server until we read simple values. It helps to disable security options.

Another important point is to do opc related stuff in a background task since doing it in main ui thread causes problems.

barnstee, 2017-02-27:
Thanks @solidrevolution, this is great!
A few comments from my side:
1. You can **use NuGet packages** instead of having to find the DLLs for the OPC UA stack and SDK. See here for an example: https://github.com/Azure/iot-gateway-opc-ua/blob/master/src/Opc.Ua.Client.Module/project.json
2. **Don't turn off security**! Or if you do, make sure you re-enable it again. I've seen too may examples of clients and servers with no secuirty because at some stage it was turned off for "easy debugging" but then never turned on again. If your client or server doesn't work with all security settings, this is a bug!
3. The sampels from the old .Net stack are all **ported to the .Net Standard stack** and in the Samples folder, no need to go back and forth.

johnn82, 2017-05-29:
Hi, I'm trying to use this library to run a simple test app on iOS with Xamarin.iOS. Does anyboady makes it work with Xamarin.iOS?

barnstee, 2017-07-17:
We've decided to contribute a Xamarin client since there is a lot of interest for it. The reason why security doesn't work is because of a limitation of Mono, used in Xamarin. We're in discussions with the Xamarin team on how to resolve this.

harpreet-singh-bola, 2017-07-22:
Great step!

Thank you.

HSB

KmanViel, 2017-07-23:
Where can this client be found?

barnstee, 2017-07-23:
It's not done yet, but when it is, it will be under "Samples".

KmanViel, 2017-07-23:
Please post here when it is ready.  This would be extremely helpful.  Any ideas on the timeline?

philipperequile, 2017-07-26:
Great!

KmanViel, 2017-08-21:
Any updates on this?

KmanViel, 2017-08-21:
I have Xamarin Forms code which works with UWP but does not work with Android.  The code builds but the following error happens during debug.

08-21 12:02:31.645 D/Mono    ( 6677): Assembly Loader probing location: 'System.Security.Principal.Windows'.
08-21 12:02:31.645 F/monodroid-assembly( 6677): Could not load assembly 'System.Security.Principal.Windows' during startup registration.
08-21 12:02:31.645 F/monodroid-assembly( 6677): This might be due to an invalid debug installation.
08-21 12:02:31.645 F/monodroid-assembly( 6677): A common cause is to 'adb install' the app directly instead of doing from the IDE.

Any ideas?

KmanViel, 2017-08-22:
It does work in Release but not in Debug.  The good news is that the program does read from and write to a UA Server using Android!

KmanViel, 2017-08-22:
To be clear I am still wanting to know if anyone has ideas about why Debug does not work.

mregen, 2017-09-20:
Blocked by Issue https://github.com/OPCFoundation/UA-.NETStandardLibrary/issues/216

mregen, 2018-02-06:
#337 is on the way...

mregen, 2018-02-08:
Please test the new UA Xamarin Client project and provide feedback or even better improvements!
Cheers!

closed 2018-02-08 by mregen

leafgreen-0312, 2021-12-27:
I have tested UA Xamarin Client project. However the following error happens when I press the Connect Button:
System.NullReferenceException
  Message=Object reference not set to an instance of an object.
I have no idea to handle this.
Any ideas? 
Many thanks!