OPCFoundation/UA-.NETStandard #57: Run Opc.Ua.Publish.Net4 with different port

lianxiwang, 2016-09-08:
Hi,

TCP 5671 port is not open in our company network and does anybody know how to configure Opc.Ua.Publisher.config.xml to allow it run through port 443, thanks in advance.

marcschier, 2016-09-08:
Hi lianxiwang,

could you try adding

`<WebSocketEndpoint>/$iothub/websocket</WebSocketEndpoint>`

to your configuration XML.  You can set the Port to 443 or just leave it out of the configuration (since 443 is used as default port when a web socket endpoint is configured).  

This should make the Publisher connect via AMQP over Websocket to IoT Hub, and provided that websockets can go through your company Firewall, should work as intended, however, if you find it not working, let me know.

Thank you!

Marc

barnstee, 2016-09-12:
Re-open if it is not working for you.

lianxiwang, 2016-09-14:
Hi Marc,
I tried your suggestion but still not working.

After updating the XML as you suggested, it ran into "return new Address(Host, Port, null, null, WebSocketEndpoint, "wss");" in GetAddress function as expected.

But it seems it hung in the code of "m_connection = await factory.CreateAsync(GetAddress());" and no return of this function, so m_closed is always true as initialized.

Appreciate your help of it, thanks in advance.

marcschier, 2016-09-19:
Hi lianxiwang,

I got a chance to try this out just now.  Publisher.net4 works as expected with AMQP over WS, but unfortunately the UWP publisher hangs.  Since it is the same code between both, it points to .net core and it would take a bit to investigate and fix.  

In the meantime, is it viable for you to use Publisher.net4 instead and see if you can make it work in your corporate infrastructure?

Thx.

Marc

barnstee, 2016-09-21:
@lianxiwang, is this working for you now?

lianxiwang, 2016-09-23:
Hi Marc,

Actually I used Pubisher.net when I encountered this issue, is it possible you can share me your Opc.Ua.Publisher.config.xml?

Thanks,
Wayne

marcschier, 2016-09-29:
Hi lianxiwang, 

I am fairly certain I did not do anything other than adding 

 `<WebSocketEndpoint>/$iothub/websocket</WebSocketEndpoint>`

(or as a matter of fact, uncomment it in the sample xml and adding my iothub credentials).  I know same config I used blocked in the UWP sample, and I believe it has to do with client cert challenge in amqplite, but this needs to be debugged by someone that knows amqplite better.  

I will try the net4 publisher one more time next week, just to make sure I did not make a mistake here.  

In the meantime, my suggestion would be to try the azure-iot-sdks C# sample from the same machine and see if it can successfully connect and send events to your iothub (it will auto fail over to websocket if it cannot connect through port 5671).  If that works it would exclude your network and security settings and point to amqplite.

Thx.

Marc

barnstee, 2016-10-05:
Closing, please reopen if this is still an issue.

closed 2016-10-05 by barnstee