OPCFoundation/UA-.NETStandard #61: Duplicate ReturnBuffer when using UA.Net Client

barnstee, 2016-09-13:
Trace with TRACE_MEMORY defined:
19:42:42.585 Channel 0 in Connecting state.
19:42:42.632 TakeBuffer(21AF1A5,10000,ReadNextMessage)
19:42:42.632 Channel 1 in Connecting state.
19:42:42.648 TCPSERVERCHANNEL SOCKET ATTACHED: 028F1359, ChannelId=1
19:42:42.663 TakeBuffer(253E9CE,10000,ReadNextMessage)
19:42:42.648 LockBuffer(21AF1A5,10000)
19:42:42.663 LockBuffer(253E9CE,10000)
19:42:42.694 Channel 0: SendHelloMessage()
19:42:42.694 TakeBuffer(10D5FB2,10000,SendHelloMessage)
19:42:42.694 Channel 2 in Connecting state.
19:42:42.710 TCPSERVERCHANNEL SOCKET ATTACHED: 014A4F1C, ChannelId=2
19:42:42.741 TakeBuffer(2308F08,10000,ReadNextMessage)
19:42:42.741 LockBuffer(2308F08,10000)
19:42:42.710 UnlockBuffer(253E9CE,10000)
19:42:42.773 LockBuffer(253E9CE,10000)
19:42:42.741 UnlockBuffer(2308F08,10000)
19:42:42.773 UnlockBuffer(253E9CE,10000)
19:42:42.804 ReturnBuffer(2308F08,10000,DoReadComplete)
19:42:42.804 TakeBuffer(2308F08,10000,ProcessHelloMessage)
19:42:42.819 Channel 1 in Opening state.
19:42:42.819 UnlockBuffer(21AF1A5,10000)
19:42:42.851 LockBuffer(21AF1A5,10000)
19:42:42.819 ReturnBuffer(253E9CE,10000,OnMessageReceived)
19:42:42.882 TakeBuffer(253E9CE,10000,ReadNextMessage)
19:42:42.882 LockBuffer(253E9CE,10000)
19:42:42.851 UnlockBuffer(21AF1A5,10000)
19:42:42.898 Channel 0: ProcessAcknowledgeMessage()
19:42:42.898 Channel 0 in Opening state.
19:42:42.913 TakeBuffer(3D8C971,10000,WriteAsymmetricMessage)
19:42:42.929 TakeBuffer(D1713E,10000,Encrypt)
19:42:42.929 ReturnBuffer(3D8C971,10000,WriteAsymmetricMessage)
19:42:42.929 UnlockBuffer(253E9CE,10000)
19:42:42.976 LockBuffer(253E9CE,10000)
19:42:42.929 ReturnBuffer(21AF1A5,10000,OnMessageReceived)
19:42:43.007 TakeBuffer(3D8C971,10000,ReadNextMessage)
19:42:43.007 ReturnBuffer(D1713E,10000,WriteOperation)
19:42:42.976 UnlockBuffer(253E9CE,10000)
19:42:43.007 LockBuffer(3D8C971,10000)
19:42:43.054 TakeBuffer(34E5D84,10000,Decrypt)
19:42:43.054 Security Policy: http://opcfoundation.org/UA/SecurityPolicy#None
19:42:43.069 Sender Certificate: (none)
19:42:43.085 SECURE CHANNEL CREATED [TcpServerChannel UA-TCP 1.02.335.0] [ID=1] Connected To: opc.tcp://erichb02:61210/UA/SampleClient [None/None/Binary]
19:42:43.085 Channel 1 in Open state.
19:42:43.085 Channel 1: SendOpenSecureChannelResponse()
19:42:43.101 TakeBuffer(2E5BEBB,10000,WriteAsymmetricMessage)
19:42:43.101 TakeBuffer(556DB3,10000,Encrypt)
19:42:43.116 ReturnBuffer(2E5BEBB,10000,WriteAsymmetricMessage)
19:42:43.116 UnlockBuffer(3D8C971,10000)
19:42:43.148 LockBuffer(3D8C971,10000)
19:42:43.116 ReturnBuffer(34E5D84,10000,ProcessOpenSecureChannelRequest)
19:42:43.179 ReturnBuffer(253E9CE,10000,OnMessageReceived)
19:42:43.179 TakeBuffer(2E5BEBB,10000,ReadNextMessage)
19:42:43.148 UnlockBuffer(3D8C971,10000)
19:42:43.210 ReturnBuffer(556DB3,10000,WriteOperation)
19:42:43.210 LockBuffer(2E5BEBB,10000)
19:42:43.241 Channel 0: ProcessOpenSecureChannelResponse()
19:42:43.257 TakeBuffer(D1713E,10000,Decrypt)
19:42:43.257 Security Policy: http://opcfoundation.org/UA/SecurityPolicy#None
19:42:43.273 Sender Certificate: (none)
19:42:43.273 SECURE CHANNEL CREATED [TcpClientChannel UA-TCP 1.02.335.0] [ID=1] Connected To: opc.tcp://localhost:61210/UA/SampleClient [None/None/Binary]
19:42:43.288 Channel 1 in Open state.
19:42:43.288 ReturnBuffer(D1713E,10000,ProcessOpenSecureChannelResponse)
19:42:43.288 ReturnBuffer(3D8C971,10000,OnMessageReceived)
19:42:43.304 TakeBuffer(D1713E,10000,ReadNextMessage)
19:42:43.304 LockBuffer(D1713E,10000)
19:42:43.304 TakeBuffer(1400C58,10000,ArraySegmentStream.Write)
19:42:43.319 TransferBuffer(1400C58,10000,WriteSymmetricMessage)
19:42:43.319 UnlockBuffer(2E5BEBB,10000)
19:42:43.351 LockBuffer(2E5BEBB,10000)
19:42:43.351 UnlockBuffer(2E5BEBB,10000)
19:42:43.319 ReturnBuffer(1400C58,10000,WriteOperation)
19:42:43.366 Channel 1: ProcessRequestMessage 1
19:42:43.382 ReturnBuffer(2E5BEBB,10000,ProcessRequestMessage)
19:42:43.382 TakeBuffer(556DB3,10000,ReadNextMessage)
19:42:43.382 LockBuffer(556DB3,10000)
19:42:43.398 Channel 1: SendResponse 1
19:42:43.398 TakeBuffer(2428E43,10000,ArraySegmentStream.Write)
19:42:43.398 TransferBuffer(2428E43,10000,WriteSymmetricMessage)
19:42:43.413 ReturnBuffer(2428E43,10000,WriteOperation)
19:42:43.413 UnlockBuffer(D1713E,10000)
19:42:43.429 LockBuffer(D1713E,10000)
19:42:43.429 UnlockBuffer(D1713E,10000)
19:42:43.444 ReturnBuffer(D1713E,10000,ProcessResponseMessage)
19:42:43.444 TakeBuffer(1400C58,10000,ReadNextMessage)
19:42:43.444 LockBuffer(1400C58,10000)
19:42:43.444 Channel 1: Close
19:42:43.460 Channel 1 in Closing state.
19:42:43.476 Channel 1: SendCloseSecureChannelRequest()
19:42:43.476 TakeBuffer(3D893C,10000,ArraySegmentStream.Write)
19:42:43.476 TransferBuffer(3D893C,10000,WriteSymmetricMessage)
19:42:43.491 UnlockBuffer(556DB3,10000)
19:42:43.491 ReturnBuffer(3D893C,10000,WriteOperation)
19:42:43.538 LockBuffer(556DB3,10000)
19:42:43.569 UnlockBuffer(556DB3,10000)
19:42:43.585 ReturnBuffer(556DB3,10000,ProcessCloseSecureChannelRequest)
19:42:43.585 TCPSERVERCHANNEL ProcessCloseSecureChannelRequest Socket=028F1359, ChannelId=1, TokenId=1
19:42:43.585 UnlockBuffer(1400C58,10000)
19:42:43.616 ReturnBuffer(1400C58,10000,DoReadComplete)
19:42:43.585 Channel 1 in Closed state.
19:42:43.632 Channel 1 closed
19:42:43.632 ReturnBuffer(556DB3,10000,OnMessageReceived)
Exception thrown: 'System.InvalidOperationException' in Opc.Ua.Core.dll
Additional information: Buffer has been locked.

barnstee, 2016-09-13:
A little more cleaned-up trace:
19:42:42.648 TCPSERVERCHANNEL SOCKET ATTACHED: 028F1359, ChannelId=1
19:42:42.694 Channel 0: SendHelloMessage()
19:42:42.694 TakeBuffer(10D5FB2,10000,SendHelloMessage)
19:42:42.694 Channel 2 in Connecting state.
19:42:42.710 TCPSERVERCHANNEL SOCKET ATTACHED: 014A4F1C, ChannelId=2
19:42:42.804 TakeBuffer(2308F08,10000,ProcessHelloMessage)
19:42:42.819 Channel 1 in Opening state.
19:42:42.898 Channel 0: ProcessAcknowledgeMessage()
19:42:42.898 Channel 0 in Opening state.
19:42:43.054 Security Policy: http://opcfoundation.org/UA/SecurityPolicy#None
19:42:43.069 Sender Certificate: (none)
19:42:43.085 SECURE CHANNEL CREATED [TcpServerChannel UA-TCP 1.02.335.0] [ID=1] Connected To: opc.tcp://erichb02:61210/UA/SampleClient [None/None/Binary]
19:42:43.085 Channel 1 in Open state.
19:42:43.085 Channel 1: SendOpenSecureChannelResponse()
19:42:43.241 Channel 0: ProcessOpenSecureChannelResponse()
19:42:43.257 Security Policy: http://opcfoundation.org/UA/SecurityPolicy#None
19:42:43.273 Sender Certificate: (none)
19:42:43.273 SECURE CHANNEL CREATED [TcpClientChannel UA-TCP 1.02.335.0] [ID=1] Connected To: opc.tcp://localhost:61210/UA/SampleClient [None/None/Binary]
19:42:43.288 Channel 1 in Open state.
19:42:43.366 Channel 1: ProcessRequestMessage 1
19:42:43.382 TakeBuffer(556DB3,10000,ReadNextMessage)
19:42:43.382 LockBuffer(556DB3,10000)
19:42:43.398 Channel 1: SendResponse 1
19:42:43.444 Channel 1: Close
19:42:43.460 Channel 1 in Closing state.
19:42:43.476 Channel 1: SendCloseSecureChannelRequest()
19:42:43.491 UnlockBuffer(556DB3,10000)
19:42:43.538 LockBuffer(556DB3,10000)
19:42:43.569 UnlockBuffer(556DB3,10000)
19:42:43.585 ReturnBuffer(556DB3,10000,ProcessCloseSecureChannelRequest)
19:42:43.585 TCPSERVERCHANNEL ProcessCloseSecureChannelRequest Socket=028F1359, ChannelId=1, TokenId=1
19:42:43.585 Channel 1 in Closed state.
19:42:43.632 Channel 1 closed
19:42:43.632 ReturnBuffer(556DB3,10000,OnMessageReceived)
Exception thrown: 'System.InvalidOperationException' in Opc.Ua.Core.dll
Additional information: Buffer has been locked.

mregen, 2016-09-14:
found duplicate buffer return --> #63

closed 2016-09-14 by barnstee