OPCFoundation/UA-.NETStandard #66: UWP apps always hit UnauthorizedAccessException

mregen, 2016-09-20:
fresh enlistment, create the certs, start the app: the certs are in a read only folder. But certs should be in a writable folder

> Adding certificate to trusted peer store. StorePath=C:\Users\xxx\Source\Repos\UA-.NETStandardLibrary\SampleApplications\Samples\Publisher\bin\x64\Debug\AppX\OPC Foundation\CertificateStores\UA Applications
Exception thrown: 'System.UnauthorizedAccessException' in System.IO.FileSystem.dll

barnstee, 2016-09-21:
UWP doesn't have write access to the app's application directory, where the directory-based. The TPM store will fix that, or we need to move the application data Directory into the app (from the OPC.UA.Core assembly).

barnstee, 2016-09-29:
Meant to say: "...where the directory-based cert store is located" above!

barnstee, 2016-10-04:
Utils.ReplaceSpecialFolderNames() is a candidate where a path replacement can be implemented.

barnstee, 2016-10-05:
The app data directory in UWP can be programmatically retrieved using Windows.Storage.ApplicationData.Current.LocalFolder.Path

mregen, 2016-12-07:
Also fixed in #94

closed 2016-12-09 by barnstee