OPCFoundation/UA-.NETStandard #70: Running on WinRT (UWP)?

tomaskrutak, 2016-10-03:
Hello, I would like to know if this OPC library is able to run as UWP application. I am trying to make the project UWP but because not all libraries are available as in full .NET framework (cryptography) then it does not work.
Thank you

barnstee, 2016-10-03:
UWP is fully supported. If you open the solution file in https://github.com/OPCFoundation/UA-.NETStandardLibrary you'll see both sample Client and Server UWP applications.
What do you mean by "not all libraries are available as in full .NET framework (cryptography)"?

tomaskrutak, 2016-10-03:
First I have to mention that I am not professional very experienced programmer. But my goal was to make running OPC client or server on raspberry Pi 3 as I understand then app must be working under WinRT UWP framework? For instance this OPC library is usinf .NET framework classes in the namespace System.Security.Cryptography.X509Certificates which is not supported by UWP application.

barnstee, 2016-10-03:
You are right, you need to run a Universal app on Windows for Raspberry Pi3. The .NEt Team did add support for X509 certificates over a year ago, see here: https://github.com/dotnet/corefx/commit/484931a343eb48d326cb698c8766a28d0e079aee

Just try this sample on your Raspi3. It works as-is:
https://github.com/OPCFoundation/UA-.NETStandardLibrary/tree/master/SampleApplications/Samples/Client
(it is a combined client/Server)

tomaskrutak, 2016-10-03:
Thank you very much for support I will try!

barnstee, 2016-10-03:
Great, let me know if you get it working...

tomaskrutak, 2016-10-03:
I downloaded, built solution and finally was able to deploy on my raspberry :). Its running.. So thanks once more!

closed 2016-10-03 by tomaskrutak