OPCFoundation/UA-.NETStandard #71: Problem running createcert.sh

MikhailKrasnobaev, 2016-10-03:
Good day,

had a problem running createcert.sh in UA-.NETStandardLibrary\SampleApplications\Samples\NetCoreConsoleClient\  in Ubuntu 14.04
Needed to run : 

 sed` -i -e 's/\r$//' createcert.sh 

Best,
Mikhail

mregen, 2016-10-04:
Thanks for the report @MikhailKrasnobaev , did you have to remove CR to be able to run the scripts? I haven't seen this problem on my cloned repo on Ubuntu 14. Could you provide repro steps?

Thanks,
Martin

mregen, 2016-10-04:
Hi @MikhailKrasnobaev, yes now it makes sense. The windows git clone adds \r\n as line ends while the linux git clone just adds the \n, so we see the different behaviour. I'd recommend to clone on Ubuntu and windows directly from your github fork and to avoid the copy to solve the issue.

barnstee, 2016-10-05:
Closing, please re-open if this is still an issue.

closed 2016-10-05 by barnstee