OPCFoundation/UA-.NETStandard #77: OPC UA Publisher sample

OscarReynhout, 2016-10-14:
I was following the tutorial to create the telemetry example, but ran into problems with the following step: _Now run the OPC UA Publisher sample, connect to a OPC UA server and publish a node._

There is no OPC UA Publisher sample in this repository! Has it been deleted or am I missing something here? Should I use the _NetCoreConsolePublisher_ ?

I want to set up a sample to understand the OPC UA and IOT Hub connection, any recommendation on how to continue?

barnstee, 2016-10-14:
NetCoreConsolePublisher is one option. It is located here:
https://github.com/OPCFoundation/UA-.NETStandardLibrary/tree/prototyping/SampleApplications/Samples/NetCoreConsolePublisher
The others are here:
https://github.com/OPCFoundation/UA-.NETStandardLibrary/tree/prototyping/SampleApplications/Samples/Publisher.Net4
https://github.com/OPCFoundation/UA-.NETStandardLibrary/tree/prototyping/SampleApplications/Samples/Publisher (UWP)

OscarReynhout, 2016-10-14:
Thanks! I was looking for the last two, but somehow I can't see them in my GitHub environment (100% new to Git :) )
[image]

OscarReynhout, 2016-10-14:
Ah, got it, master vs prototyping. So what remains of this issue is that the tutorial does not match the master, but I can continue. Thanks!

barnstee, 2016-10-14:
Sorry, yes, the publishers will go back to master once the Pub/Sub spec is officially released.

closed 2016-10-14 by barnstee