OPCFoundation/UA-.NETStandard #82: Xamarin.form OPC UA Client session connection to OPC UA Server

CarlCrol, 2016-10-24:
Connecting to an OPC UA Server through an Android App developed using Xamarin.form is not successful .
In Session.Connect, it throws an exception, connection refused...

mregen, 2016-11-03:
@CarlCrol Could you provide the exception call stack and more info on the exception?

CarlCrol, 2016-11-03:
Okay will send tomorrow. My dev machine is currently updating

CarlCrol, 2016-11-08:
[image]

CarlCrol, 2016-11-08:
The error is an System.ComponentModel.Win32Exception
Message is "Could not resolve host 'xxxxx'" 
SocketErrorCode: System.Net.Sockets.SocketError.HostNotFound

CarlCrol, 2016-11-08:
Here is the error message:
Exception Type:
System.TypeLoadException
Error message:
"Could not load type 'System.Reflection.TypeExtensions' from assembly 'System.Reflection.TypeExtensio…"
StackTrace:
"  at Opc.Ua.ServiceResult.LookupSymbolicId (UInt32 code) [0x00001] in C:\\Users\\carlcrol\\Downloads\\UA…"

CarlCrol, 2016-11-08:
Please advice and if you have sample code using OPC UA SDK in Xamarin.Form

CarlCrol, 2016-11-08:
Here is the error:

>>> An unexpected error occurred while connecting to the server.

Additional Error:
---   at Opc.Ua.Bindings.TcpAsyncOperation`1[T].End (Int32 timeout) [0x00091] in C:\Users\carlcrol\Downloads\UA-.NETStandardLibrary-master\UA-.NETStandardLibrary-master\Stack\Opc.Ua.Core\Stack\Tcp\TcpAsyncOperation.cs:172 
---   at Opc.Ua.Bindings.TcpClientChannel.OnConnectOnDemandComplete (System.Object state) [0x0003d] in C:\Users\carlcrol\Downloads\UA-.NETStandardLibrary-master\UA-.NETStandardLibrary-master\Stack\Opc.Ua.Core\Stack\Tcp\TcpClientChannel.cs:1200 

Localized Text:
Error establishing a connection: An unexpected error occurred while connecting to the server.

StackTrace:
  at Opc.Ua.Bindings.TcpAsyncOperation`1[T].End (Int32 timeout) [0x00091] in C:\Users\carlcrol\Downloads\UA-.NETStandardLibrary-master\UA-.NETStandardLibrary-master\Stack\Opc.Ua.Core\Stack\Tcp\TcpAsyncOperation.cs:172 
  at Opc.Ua.Bindings.TcpClientChannel.EndSendRequest (IAsyncResult result) [0x0001e] in C:\Users\carlcrol\Downloads\UA-.NETStandardLibrary-master\UA-.NETStandardLibrary-master\Stack\Opc.Ua.Core\Stack\Tcp\TcpClientChannel.cs:305 
  at Opc.Ua.Bindings.TcpTransportChannel.EndSendRequest (IAsyncResult result) [0x00026] in C:\Users\carlcrol\Downloads\UA-.NETStandardLibrary-master\UA-.NETStandardLibrary-master\Stack\Opc.Ua.Core\Stack\Tcp\TcpTransportChannel.cs:322 
  at Opc.Ua.Bindings.TcpTransportChannel.SendRequest (IServiceRequest request) [0x0000b] in C:\Users\carlcrol\Downloads\UA-.NETStandardLibrary-master\UA-.NETStandardLibrary-master\Stack\Opc.Ua.Core\Stack\Tcp\TcpTransportChannel.cs:272 
  at Opc.Ua.DiscoveryClient.GetEndpoints (Opc.Ua.RequestHeader requestHeader, System.String endpointUrl, Opc.Ua.StringCollection localeIds, Opc.Ua.StringCollection profileUris, Opc.Ua.EndpointDescriptionCollection& endpoints) [0x00047] in C:\Users\carlcrol\Downloads\UA-.NETStandardLibrary-master\UA-.NETStandardLibrary-master\Stack\Opc.Ua.Core\Stack\Generated\Opc.Ua.Client.cs:4827 
  at Opc.Ua.DiscoveryClient.GetEndpoints (Opc.Ua.StringCollection profileUris) [0x00003] in C:\Users\carlcrol\Downloads\UA-.NETStandardLibrary-master\UA-.NETStandardLibrary-master\Stack\Opc.Ua.Core\Stack\Client\DiscoveryClient.cs:68 
  at OpcUaMobileSignalR.SessionManager.SelectEndpoint (System.String discoveryUrl, Boolean useSecurity) [0x0005d] in C:\Files\Developer\OPC UA\OpcUaDevelopment\OpcUaMobileSignalR\OpcUaMobileSignalR\Session\SessionManager.cs:269 
  at OpcUaMobileSignalR.SessionManager.CreateSessionImplementation (System.String url) [0x00147] in C:\Files\Developer\OPC UA\OpcUaDevelopment\OpcUaMobileSignalR\OpcUaMobileSignalR\Session\SessionManager.cs:212 
  at OpcUaMobileSignalR.SessionManager.CreateSession (System.String url, System.String connectionId) [0x0004d] in C:\Files\Developer\OPC UA\OpcUaDevelopment\OpcUaMobileSignalR\OpcUaMobileSignalR\Session\SessionManager.cs:68 
  at OpcUaMobileSignalR.App.OnStart () [0x00002] in C:\Files\Developer\OPC UA\OpcUaDevelopment\OpcUaMobileSignalR\OpcUaMobileSignalR\App.cs:33

CarlCrol, 2016-11-08:
I copied the code to a WPF test application that i used in Xamarin.Form.
It worked for WPF Test application but still fails in Xamarin.Form

pedosoul, 2016-11-16:
CarlCrol: I would like to contribute on this project to use OPC UA on Xamarin forms,

Do you have some news about your issue ?

How do you use the UA.NET standardLibrary into your Mobile Project ? 
For information i'm using vs2015 with xamarin.

Perhaps, Could you provide me your project for trying to work it ?

Thanks.

pedosoul, 2016-11-16:
@CarlCrol i send u an email, if you can reply me some info about how to get OPC UA client on Xamarin Android

CarlCrol, 2016-11-17:
I've attached the sample project please take a look. 

https://www.dropbox.com/s/k3wppe5eqrbb77h/OpcUa.zip?dl=0

tobiasstraub, 2016-11-25:
@CarlCrol @pedosoul I ported the UA Stack to Xamarin as part of my Bachelor thesis. If there is still an interest, I can upload my source code. My thesis will be published in the next few weeks.

pedosoul, 2016-11-26:
Hi,

Of course its interesting  !

Ismaël

CarlCrol, 2016-11-26:
Yes please. can you send it need it

MikhailKrasnobaev, 2016-11-30:
@tobiasstraub Hi, will you publish the code somewhere?

tobiasstraub, 2016-12-02:
@MikhailKrasnobaev I'm talking at the moment with the company where I wrote my thesis. So it can take a while, but as soon as I have the green light, I will publish the source code. By the way, I am writing at the moment on a [Xamarin UA Skeleton](http://ts.digital/ua-xamarin-skeleton). With this, the start for own Xamarin + UA projects should be easier.

If you're interested, you can [subscribe via E-Mail to updates for the Skeleton](http://ts.digital/ua-xamarin-skeleton) and Thesis source code publishing state.

barnstee, 2016-12-02:
That's good news, Tobias, would love to see you contribute a Xamarin client to this repo!

pedosoul, 2016-12-02:
I'm waiting for the solution of tobias too !  I suscribe to his mailing.
I'm working for a Siemens customer which design milling machine and I need
to developpe a android application using opc ua.
Thanks in advance.
Ismael

pedosoul, 2016-12-07:
Hi all,
I'm still blocked on the last error explain on my email.
I hope tobias will be autorized to send us his solution.
During this time could weexplain how/why do you need to used the OPC UA
Xamarin.

For me, I'm working for Fives machining and I study a solution on Android
to use communicate with an Siemens 840Dsl Command Numeric (CNC)

Ismaël

awcullen, 2016-12-13:
I invite all to try the 'Workstation.UaClient' library on Nuget.  Supports Xamarin Forms (MonoAndroid)  as of version 1.5
See the samples at https://github.com/convertersystems/opc-ua-samples
Any feedback is most welcome!

Andrew

pedosoul, 2016-12-13:
Thanks I'll try it soon !

CarlCrol, 2016-12-21:
Where you able to make it work? I tried doing a sample application but in this line of code.
private string discoveryUrl = @"opc.tcp://10.164.232.82:62547"; // Use ip address ( or '10.0.2.2' for accessing local host computer from emulator. hostname, localhost or 127.0.0.1 will not work on Android emu!)
        private ILoggerFactory loggerFactory;        private ILogger logger;        private UaTcpSessionClient session;        protected override void OnCreate (Bundle bundle) { base.OnCreate (bundle);
 global::Xamarin.Forms.Forms.Init (this, bundle);            // Setup a logger.            this.loggerFactory = new LoggerFactory();              this.logger = this.loggerFactory.CreateLogger<MainActivity>();
            try            {                // Create the session client for the app.                this.session = new UaTcpSessionClient(                    new ApplicationDescription()                    {                        ApplicationName = "Workstation.MobileHmi",                        ApplicationUri = $"urn:{System.Net.Dns.GetHostName()}:Workstation.MobileHmi",                        ApplicationType = ApplicationType.Client                    },                    new DirectoryStore(                        Path.Combine(System.Environment.GetFolderPath(System.Environment.SpecialFolder.LocalApplicationData), "pki"),                        loggerFactory: loggerFactory),                    this.ProvideUserIdentity,                    this.discoveryUrl,                    this.loggerFactory);
                this.LoadApplication(new App(this.session));            }            catch (Exception ex)            {                this.logger.LogError("Error creating UaTcpSessionClient. {0}", ex.Message);            }        }

ex:{System.TypeInitializationException: The type initializer for 'System.Collections.Generic.List`1' threw an exception. ---> System.TypeLoadException: Could not load type 'Workstation.ServiceModel.Ua.Subscription[]' from assembly 'Workstation.UaClient, Version=1.5.0.30692, Culture=neutral, PublicKeyToken=null'.  at System.Collections.Generic.List`1[T]..cctor () [0x00000] in /Users/builder/data/lanes/3415/7db2aac3/source/mono/external/referencesource/mscorlib/system/collections/generic/list.cs:47   --- End of inner exception stack trace ---  at (wrapper managed-to-native) System.Object:__icall_wrapper_mono_generic_class_init (intptr)  at System.Collections.ObjectModel.Collection`1[T]..ctor () [0x00006] in /Users/builder/data/lanes/3415/7db2aac3/source/mono/external/referencesource/mscorlib/system/collections/objectmodel/collection.cs:28   at System.Collections.ObjectModel.ObservableCollection`1[T]..ctor () [0x0000b] in /Users/builder/data/lanes/3415/7db2aac3/source/mono/external/referencesource/System/compmod/system/collections/objectmodel/observablecollection.cs:50   at Workstation.ServiceModel.Ua.UaTcpSessionClient..ctor (Workstation.ServiceModel.Ua.ApplicationDescription localDescription, ICertificateStore certificateStore, System.Func`2 userIdentityProvider, System.String endpointUrl, ILoggerFactory loggerFactory, Double sessionTimeout, UInt32 timeoutHint, UInt32 diagnosticsHint, UInt32 localReceiveBufferSize, UInt32 localSendBufferSize, UInt32 localMaxMessageSize, UInt32 localMaxChunkCount) [0x0002b] in <filename unknown>:0   at App1.Droid.MainActivity.OnCreate (Android.OS.Bundle bundle) [0x0002e] in D:\Files\Personal\Development\OpcUa\App1\App1\App1.Droid\MainActivity.cs:37 }

stacktrace:  at (wrapper managed-to-native) System.Object:__icall_wrapper_mono_generic_class_init (intptr)  at System.Collections.ObjectModel.Collection`1[T]..ctor () [0x00006] in /Users/builder/data/lanes/3415/7db2aac3/source/mono/external/referencesource/mscorlib/system/collections/objectmodel/collection.cs:28   at System.Collections.ObjectModel.ObservableCollection`1[T]..ctor () [0x0000b] in /Users/builder/data/lanes/3415/7db2aac3/source/mono/external/referencesource/System/compmod/system/collections/objectmodel/observablecollection.cs:50   at Workstation.ServiceModel.Ua.UaTcpSessionClient..ctor (Workstation.ServiceModel.Ua.ApplicationDescription localDescription, ICertificateStore certificateStore, System.Func`2 userIdentityProvider, System.String endpointUrl, ILoggerFactory loggerFactory, Double sessionTimeout, UInt32 timeoutHint, UInt32 diagnosticsHint, UInt32 localReceiveBufferSize, UInt32 localSendBufferSize, UInt32 localMaxMessageSize, UInt32 localMaxChunkCount) [0x0002b] in <filename unknown>:0   at App1.Droid.MainActivity.OnCreate (Android.OS.Bundle bundle) [0x0002e] in D:\Files\Personal\Development\OpcUa\App1\App1\App1.Droid\MainActivity.cs:37

awcullen, 2016-12-22:
Carl,
I've provided some details at my repo https://github.com/convertersystems/opc-ua-samples/issues/6
Andrew

Savoyyang, 2017-04-21:
@CarlCrol @tobiasstraub   Did you continue to update the code use .NETStandard later ?

tobiasstraub, 2017-04-21:
Hi @Savoyyang, from my side, the UA-.NETStandardLibrary can be used within a Xamarin application, but the UA Xamarin sample itself is not up to date with the changes [Xamarin introduces at the end of 2016](https://blog.xamarin.com/net-standard-library-support-for-xamarin/) (sample itself is not a StandardLibrary). If there is an interest, I update the sample as fast my calendar allows it.

Savoyyang, 2017-04-21:
@tobiasstraub  yes , Don't know if you have seen the https://github.com/OPCFoundation/UA-.NETStandardLibrary/issues/54 ,   I follow the issue and completed the built environment, can be build in Xamarin. And I reference NetCoreConsoleClient code , but fail at "Create an Application Configuration” with security. I don't konw how to solution it, So looking forward to your sample.Please remember @me.
Thank you for you contribution.

barnstee, 2017-07-17:
Closing as a duplicate of https://github.com/OPCFoundation/UA-.NETStandardLibrary/issues/54

closed 2017-07-17 by barnstee