OPCFoundation/UA-.NETStandard #88: Cannot connect to UA Server in secure mode using policy Basic256Sha256

RavilN, 2016-11-29:
When this library is used by the iot-gateway-opc-ua project, the client cannot connect to the C++ demo server from Unified Automation. The client selects endpoint with the highest security level, which uses policy Basic256Sha256, but open secure channel fails with BadInvalidArgument error message with reason [OpcUa Null String]. Can see by Wireshark  that ClientNonce is null in the request, looks like that is the cause of the problem.

mregen, 2016-11-30:
Hi @RavilN, thanks for the report. That's a bug in the gateway client. This OPC UA ,Net library doesn't support the Sha256 level (yet), so the gateway client should choose the highest supported client level.

barnstee, 2016-12-05:
Fixed via CL https://github.com/OPCFoundation/UA-.NETStandardLibrary/commit/a62d11c3fb7bb0e179404496a334860481a0863f. New NuGet package V0.0.8 will be available in iot-gateway-opc-ua today..

closed 2016-12-05 by barnstee

eltigre4jc, 2018-07-13:
Hello! I think this issue is still not solved.
See [https://github.com/OPCFoundation/UA-.NETStandard/issues/494](url)