OPCFoundation/UA-.NETStandard #97: NetCoreConsoleServer generates BadCertificateUriInvalid when NetCoreConsoleClient runs on same machine

abrasat, 2016-12-13:
When both NetCoreConsole cliend and Server applications run on same machine, the certificate check in the CreateSession() method in StandardServer.cs fails at this check:

`if (!String.IsNullOrEmpty(certificateApplicationUri) &&
 !String.IsNullOrEmpty(clientDescription.ApplicationUri) &&
 certificateApplicationUri != clientDescription.ApplicationUri)`

The certificateApplicationUri string contains "urn:MyPCName:OPCFoundation:SampleClient" and the clientDescription.ApplicationUri string contains "urn:localhost:OPCFoundation:SampleClient", and thus the string comparison fails.

barnstee, 2016-12-14:
Good catch. This could have something to do with a recent change to replace localhost in the certs with the host name of the machine. We'll investigate...

mregen, 2016-12-14:
@abrasat Hi, I cannot repro the failing check here. The sample apps have all been updated to replace localhost with the hostname if necessary and I just double checked this is working in the current master branch. Maybe there is still an old cert being used in your repo. Please delete the current certs in *OPC Foundation\CertificateStores\MachineDefault* in both projects and try again if it is still failing.

abrasat, 2016-12-14:
I have checked out the master branch again, and now is indeed working. I must have checked out the samples shortly before the fix was made.
Thanks for the help.

closed 2016-12-14 by abrasat

mregen, 2016-12-14:
@abrasat np, thanks for testing!