OPCFoundation/UA-.NETStandard #103: When connection to OPC Server without client certificate in .net core 1.1 the CertificateFactory crash

henrikstromberg, 2016-12-21:
NetCoreConsoleClient  crashes when updating it .Net core 1.1 and remove the certificates.

I´m buiding a OPC UA Client and use the possibility to connect without a certificate. it work fine util I updated my client to .net core 1.1. Now the method below crashes.

public static X509Certificate2 Create(byte[] encodedData, bool useCache)
{
}

Steps to reproduce.
 
1. Update the project.json to .net core 1.1 in  NetCoreConsoleClient 
2. Delete CertificateStores
3. update code to not use CertificateStores
     bool haveAppCertificate = false;// config.SecurityConfiguration.ApplicationCertificate.Certificate != null;

        /*    if (!haveAppCertificate)
            {
                Console.WriteLine("    INFO: Creating new application certificate: {0}", config.ApplicationName);

                X509Certificate2 certificate = CertificateFactory.CreateCertificate(
                    config.SecurityConfiguration.ApplicationCertificate.StoreType,
                    config.SecurityConfiguration.ApplicationCertificate.StorePath,
                    config.ApplicationUri,
                    config.ApplicationName
                    );

                config.SecurityConfiguration.ApplicationCertificate.Certificate = certificate;

            }

            haveAppCertificate = config.SecurityConfiguration.ApplicationCertificate.Certificate != null;
            */

barnstee, 2016-12-21:
Don't upgrade your project. Please ONLY use the combination of NuGet package versions we have tested, there are a lot of dependencies to work through. Was there a reason why you need the V1.1 package?

henrikstromberg, 2016-12-21:
Other parts of my application needs EntityFrameworkCore 1.1 and other package that that depends on .Net Core 1.1.
It seems like if I change the CertificateFactory to create a X509Certificate2() it connects and communicates with my opc server.
We will test to run it as .Net core 1.1 and validate it it works for our needs before we rewrite the rest of our system.

At the moment I reference the code directly and build the code so changing the CertificateFactory is not a problem.

Do you have a plan to go up to .net 1.1 ?

Best regards Henrik

mregen, 2017-01-12:
@henrikstromberg I did a repro and as you reported, it appears to be a problem of the .Net Core 1.1 runtime only.

https://github.com/dotnet/corefx/issues/13814
https://github.com/dotnet/coreclr/issues/8148

We need to investigate a fix/workaround. Thanks for reporting the topic.

mregen, 2017-01-21:
@henrikstromberg. There is no workaround but to wait for .Net Core 1.2 or to set this in your enviroment:

set COMPlus_ReadyToRunExcludeList=System.Security.Cryptography.X509Certificates

I provided also a launchsettings.json in Cert fixes #119.

Hope this helps.

barnstee, 2017-01-22:
Aparently, they will fix this in .NetCore 1.1.1.

mregen, 2017-01-23:
Indeed: https://github.com/dotnet/core/blob/master/release-notes/1.1/1.1-known-issues.md

barnstee, 2017-01-30:
Closing. Please reopen if the workaround doesn't work for you.

closed 2017-01-30 by barnstee