OPCFoundation/UA-.NETStandard #113: Open the Project in VS15 on Windows 7 without Windows 10 SDK

tomsoftware, 2017-01-11:
How to open the UA-.NETStandardLibrary in Visual Studio 15 on Windows 7 when Windows 10 SDK is required?

After open the "UA Sample Applications.sln" I got
[image]
but I'm not able to install the Win 10 SDK on my Windows 7 machine. 
- How to modify the project?
- What dependencys do you use from the Win 10 SDK?

mregen, 2017-01-11:
@tomsoftware You can safely delete the projects above from your solution, those are sample apps / libs for UWP and not required for .Net 4.6 or .Net core development.

tomsoftware, 2017-01-12:
Thanks @mregen,
No, I was likewise unable to open _Opc.Ua.Core.xproj_ with the errror: unable to load 
$(VSToolsPath)\DotNet\Microsoft.DotNet.targets
$(VSToolsPath)\DotNet\Microsoft.DotNet.Props

I fixed the problem by installing _Visual Studio 2015 Tools (Preview 2)_
https://www.microsoft.com/net/download/core

Thanks

closed 2017-01-12 by tomsoftware

mregen, 2017-01-12:
@tomsoftware Thanks, I will update the docs..