OPCFoundation/UA-.NETStandard #117: about Opc.Ua.Client.Control

Savoyyang, 2017-01-16:
Hello everyone , My IDE is VS2015 update 3 , win 7 and win10SDK 10586， but I open this Project and rebuild it ,in Opc.Ua.Client.Control(Universal Windows),
error code :
         The type 'IAsyncAction' is defined in an assembly that is not referenced. You must add a reference to assembly 'Windows, Version=255.255.255.255, Culture=neutral, PublicKeyToken=null, ContentType=WindowsRuntime'.

The type 'IAsyncActionWithProgress<>' is defined in an assembly that is not referenced. You must add a reference to assembly 'Windows, Version=255.255.255.255, Culture=neutral, PublicKeyToken=null, ContentType=WindowsRuntime'.

The type 'IAsyncOperation<>' is defined in an assembly that is not referenced. You must add a reference to assembly 'Windows, Version=255.255.255.255, Culture=neutral, PublicKeyToken=null, ContentType=WindowsRuntime'.

The type 'IAsyncOperationWithProgress<,>' is defined in an assembly that is not referenced. You must add a reference to assembly 'Windows, Version=255.255.255.255, Culture=neutral, PublicKeyToken=null, ContentType=WindowsRuntime'

How to solve this question? thank you very much!!!

barnstee, 2017-01-16:
Never seen this before. Folks on stackoverflow who have reported this issue said that simply recomiling the project made this error go away. The error code message is bugus anyway, as the reference is incorrect:  Version=255.255.255.255, Culture=neutral, PublicKeyToken=null doesn't exist.

Savoyyang, 2017-01-17:
morning barnstee,
         I forget told you the code in DiscoveredServerListCtrl.xaml.cs(Line 195-300) and ConfiguredServerDlg.xaml.cs(Line 1104-1115   and 1122-1133)

`        /// <summary>
        /// Updates the list of servers displayed in the control.
        /// </summary>
        private async void OnUpdateServers(object state)
        {
            if (!Dispatcher.HasThreadAccess)
            {
                await Dispatcher.RunAsync(
                CoreDispatcherPriority.Normal,
                () =>
                {
                    OnUpdateServers(state);
                });
                return;
            }`

`        /// <summary>
        /// Updates the list of servers displayed in the control.
        /// </summary>
        private async void OnUpdateServers(object state)
        {
            if (!Dispatcher.HasThreadAccess)
            {
*********************error***************************
                await Dispatcher.RunAsync(
                CoreDispatcherPriority.Normal,
                () =>
                {
                    OnUpdateServers(state);
                });
*********************error***************************
                return;
            }
`

barnstee, 2017-01-18:
Yeah, there are other places in the code where this is used, so I'm still unclear why you are seeing this there only. I always throught cross-compiling from Windows 7 for Windows 10 was not supported? See here:
https://developer.microsoft.com/en-US/windows/downloads/windows-10-sdk
Supported operating systems
Windows 10
Windows 8.1
Windows Server 2012 R2
Windows Server 2012

barnstee, 2017-01-19:
Closing, reopen if still an issue.

closed 2017-01-19 by barnstee