OPCFoundation/UA-.NETStandard #120: about OPC UA Web Telemetry sample question

Savoyyang, 2017-01-22:
morning sir ,

I read <How to build and run the OPC UA Web Telemetry sample>, but I still don't know how OpcUaWebTelemetry  works. 
Dont know {StorageAccountName}{AccessKey}{ConsumerGroupName} those parameter how to configure in Azure ...
Can someone demonstrate how OpcUaWebTelemetry connect Azure and OpcUA client/server use youtube or PDF(use picture)？
then @me , thank you very much!!!

Best Regards！

barnstee, 2017-01-23:
Yeah, the docs are not great. I just updated them to make this easier:
https://github.com/OPCFoundation/UA-.NETStandardLibrary/pull/124

Savoyyang, 2017-01-23:
If can complete a little would be great.
OPC UA Server <-> Azure Iot <-> OPC UA Web Telemetry/OPC UA Client ， Right？

barnstee, 2017-01-23:
No.:
OPC UA Server->OPC Publisher or Azure IoT Gateway SDK->IoT Hub-> OPC UA Web Telemetry :-)
OPC Publishers are here:
https://github.com/OPCFoundation/UA-.NETStandardLibrary/tree/prototyping/SampleApplications/Samples
Azure IoT Gateway SDK is here:
https://github.com/Azure/iot-gateway-opc-ua

Savoyyang, 2017-01-23:
oh , I think too simple :-)  , thank you very much!!!

closed 2017-01-27 by barnstee

tejasklodha, 2017-03-16:
Dear barnstee ,

I successfully setup OpcUaWebTelemetry and deploy to azure as below link
http://fmiiotwebapp.azurewebsites.net/

Now what is next step for me?
How my own OPC UA server connect with azure? 
where my OPC US Server define in code?

When Tried to download full UA-.NETStandardLibrary then **OPC Publishers** is not included in zip then How can I use OPC UA Publishers 

When Tried load Azure IoT Gateway SDK then project not loaded and give attached error. 
Dear barnstee ,

I successfully setup OpcUaWebTelemetry and deploy to azure as below link
http://fmiiotwebapp.azurewebsites.net/

Now what is next step for me?
How my own OPC UA server connect with azure? 
where my OPC US Server define in code?

When Tried to download full UA-.NETStandardLibrary then **OPC Publishers** is not included in zip then How can I use OPC UA Publishers 

When Tried load Azure IoT Gateway SDK then project not loaded and give attached error. 

Please guide me that what is next step? 

[image]
[image]

Savoyyang, 2017-03-17:
Wow, that's so cool, but I don't know how use it, so dispirited.

tejasklodha, 2017-03-17:
Problem resolved....

Now I am able to publish data from my opc server to Azure... :)

Savoyyang, 2017-03-17:
@tejasklodha     You are amazing,  Can you write a detailed tutorial for user reference, please
Thank you very much.

Savoyyang, 2017-03-17:
@tejasklodha     my eamil  :  yangshunfu_@live.com

tejasklodha, 2017-03-20:
Dear Savoyyang ,

What you need my help?.
I am happy to help you...

Thanks,
Tejas Lodha

Savoyyang, 2017-03-20:
Dear   Tejas Lodha,

      My purpose just like you, subscription  OPC UA server node in Azure , my tests failed before, but you made it, so I want to request you to share of your success tutorial, as detailed as possible,also share it with others , because I'm not familiar with it.

thank you very much!

tejasklodha, 2017-03-20:
Dear Savoyyang ,

Sure , Right now I have not detail tutorials but In this weekend i will make it and send to you by Email.
For sharing to others, I must taking permission to owner of this code (UA-.NETStandardLibrary) to share detail tutorials to others in public by blogs

Thanks,

Savoyyang, 2017-03-20:
Dear Tejas Lodha,

Thanks for your hard work, Looking forward to your tutorials.

thank you very much!

Savoyyang, 2017-04-05:
@tejasklodha
     Hello  tejasklodha , did you complete?Can you send to me ,  my eamil : yangshunfu_@live.com