OPCFoundation/UA-.NETStandard #130: Add an Issue/PR template

WilliamBerryiii, 2017-02-01:
Can we add an Issue/PR template to the repository to standardize how stuff is reported?  

Like this: https://github.com/blog/2111-issue-and-pull-request-templates.  

Here is an example from another project: https://github.com/fsprojects/ProjectScaffold/blob/master/.github/ISSUE_TEMPLATE.md

henrikstromberg, 2017-02-01:
Hi Bill,

Thanks for your reply. We found out that or call the database took to long time. We fixed that and runs new test tonight.
I open a issue tomorrow if we still se the problem.

Best regards Henrik

barnstee, 2017-02-01:
I don't see the need right now since free text has worked best for folks. Feel free to use a template, though...

closed 2017-02-01 by barnstee