OPCFoundation/UA-.NETStandard #131: Implementation of SetTriggering Service

mattse76, 2017-02-05:
Hello,

will the "SetTriggering" Service be implemented in the Stack ( Opc Ua, Part4: 5.12.5) 
or anything equivalent to have notifications being triggered by the change of the value of 
a variable? It would be of great importance for us ...

Greetings and thanks in advance,

Matthias

barnstee, 2017-02-06:
This is already implemented. Try any client and create a subscription with MonitoringMode.Reporting.

mattse76, 2017-02-06:
Sorry, but the “Triggering Service” is not implemented.

At least, I cannot find any point where I can set monitored items to trigger other,
like it is implemented here:

http://documentation.unified-automation.com/uasdkcpp/1.5.0/html/classUaSubscription.html#ad0c7f9d6006a8be5a87f94f642749cfa

Kind Regards,

Matthias

From: Erich Barnstedt 
Sent: Monday, February 6, 2017 8:47 AM
To: OPCFoundation/UA-.NETStandardLibrary 
Cc: mattse76 ; Author 
Subject: Re: [OPCFoundation/UA-.NETStandardLibrary] Implementation of SetTriggering Service (#131)

This is already implemented. Try any client and create a subscription with MonitoringMode.Reporting.

barnstee, 2017-02-06:
Can you please try it out? We're just not calling it the same as the UA SDK. MonitoringMode.Reporting does exactly what you are asking for, in my opinion. If it doesn't, please come back and we can discuss.

mattse76, 2017-02-08:
Hi Erich,

I think of the following scenario:

While a control recipe is being processed, 
the steps and thus the step number change. 
The step numbers may not be consecutive.
At the end of each step, a counter value (temperature, pressure) has to be read (for energy balance etc.).
At the beginning of each new step, the counter has to be reset.

When a change in step is reported, 
the corresponding counter value has to be reported, too.
However, at the time a change in step number is reported,
the counter will be reset. 

As a consequence, from the client perspective, step number 
and the process value at the end/beginning of each step will never be consistent.

I think this is a very common problem when programming 
Batch Protocols or MES/MIS Systems. 

Of course, I can create new Opc Ua Events.
I could delay the reset of the counter ...
But for this I have to change the server program.
On the client side, I am not always in the situation I can do that.

I think, a consistent solely client based solution is needed for that 
and that the “Set Triggering” service is intended for this purpose:

A triggering Item and other items get linked to this item ( by setting a vector of item Ids )
and thus forced to report when the triggering item reports (in the same notification).

However, I think even “Set Triggering” is not precise enough.
Normally one would need “OnBeforeChange” (and “OnAfterChange”) for this purpose.

What do you think?

Thank you for taking your time!

Kind Regards,

Matthias

barnstee, 2017-02-08:
Leave a comment

AlinMoldovean, 2017-09-28:
Hello @mattse76,

The SetTrigggering service support is implemented in the .NET Standard stack for both Client and Server components.
I put a sample code that can be used in the client to create triggering links for a triggering item.
However, the triggering mechanism will is not guaranteed to provide consistent data for your use-case.
A more suitable way would be to use structured data types for packing multiple values.

                // having a Session and a Subscription created
                // and the monitored items
                // make sure to set the MonitoringMode of the linked items to MonitoringMode.Sampling;

                Session session;
                Subscription subscription;

                MonitoredItem triggeringItemId;
                MonitoredItem linkedMonitoredItem1;
                MonitoredItem linkedMonitoredItem2;                

                UInt32Collection linksToAdd = new UInt32Collection(3);
                linksToAdd.Add(linkedMonitoredItem1.Status.Id);
                linksToAdd.Add(linkedMonitoredItem2.Status.Id);

                StatusCodeCollection statusCodes;
                DiagnosticInfoCollection diagnosticInfos;
                StatusCodeCollection statusCodesRemoved;
                DiagnosticInfoCollection diagnosticInfosRemoved;
                ResponseHeader responseHeader = session.SetTriggering(null, subscription.Id, triggeringItemId.Status.Id, linksToAdd, null,
                        out statusCodes,
                        out diagnosticInfos,
                        out statusCodesRemoved,
                        out diagnosticInfosRemoved);

closed 2017-10-12 by AlinMoldovean