OPCFoundation/UA-.NETStandard #133: Add support for complete distinguished subject names in configuration and to create certs

mregen, 2017-02-07:
Currently SubjectName holds only CommonName and created certs contain e.g.` CN=MES, DC=MACHINENAME` as Subject. 

However for some applications we may need to create certs with 
a complete distinguished subject name, e.g.

`<SubjectName>CN=Station/O=Contoso/OU=Lehel/C=Germany/ST=Bavaria/L=Munich/DC=localhost</SubjectName>`

we should maintain backward compatibiility with old behaviour, aka if there is no CN= the entry is treated as common name

```
The subject for the certificate 
           Note that subject names are complex structures. The text that appears here is the CommonName component.
           A complete distinguished would be something like: 'CN=MES, DC=MACHINENAME'
           The first certificate found is used if multiple certificates with the same CommonName exist. 
           The Thumbprint should be specified if the CommonName does not uniquely identify a certificate. -->

```

mregen, 2017-04-26:
working on it..

barnstee, 2017-05-07:
fixed.

closed 2017-05-07 by barnstee