OPCFoundation/UA-.NETStandard #134: Compatibility with Xamarin - Android?

philipperequile, 2017-02-13:
I am trying to get this library working on Android with Xamarin.
First I got a minimal OPC UA client working in a simple Windows forms app.
Then I tried to get it working on Android.
I can catch the server certificate file and store it in the android app personal folder.
And I can use the DiscoveryClient to get the available endpoints.
But I encounter a few problems in android.
1) CertificateFactory.CreateCertificate
-> Error in CertificateFactory.cs in X509Certificate2 CreateCertificateFromPKCS12  line 356:
		InnerException	{System.NotSupportedException}	System.NotSupportedException
		Message	"Input data cannot be coded as a valid certificate."	string
		Source	"mscorlib"	string
		StackTrace	"  at Mono.Security.X509.X509Certificate.Parse (System.Byte[] data) [0x00352] in /Users/builder/data/…"	string
2) timeout error in session start
  = await Session.Create(config, endpoint, false, this.ApplicationName,  60000, identity, null);
=> E/mono    (20031): Opc.Ua.ServiceResultException: BadRequestTimeout
   E/mono-rt (20031): [ERROR] FATAL UNHANDLED EXCEPTION: Opc.Ua.ServiceResultException: BadRequestTimeout 
3) Utils.GetHostName 
  returns "localhost". In a winforms app I get the PC hostname.

Savoyyang, 2017-04-21:
@philipperequile 
hello philipperequile ,
I made reference to this issues https://github.com/OPCFoundation/UA-.NETStandardLibrary/issues/54,
met with you on security issues, about "Create an Application Configuration", 
I think this is because Utils.GetHostName not compatibility in Android like Computer and so on .
may I ask how did you solve it?

barnstee, 2017-07-17:
Closing as duplicate of https://github.com/OPCFoundation/UA-.NETStandardLibrary/issues/54.

closed 2017-07-17 by barnstee