OPCFoundation/UA-.NETStandard #149: Fails to build with VisualStudio 2017

JuergenKosel, 2017-03-22:
Hello,

since https://github.com/Azure/iot-gateway-opc-ua-proxy requires building with Visual Studio 2017, I have updated to Visual Studio 2017.
But with Visaul Studio 2017, the build of this UA-.NETStandardLibrary fails.
The following is the build output for the first failed target:

`1>C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\MSBuild\Sdks\Microsoft.NET.Sdk\build\Microsoft.NET.TargetFrameworkInference.targets(84,5): error : Cannot infer TargetFrameworkIdentifier and/or TargetFrameworkVersion from TargetFramework='uap10.0'. They must be specified explicitly.`
`1>C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\MSBuild\15.0\Bin\Microsoft.Common.CurrentVersion.targets(1111,5): error MSB3644: The reference assemblies for framework ".NETFramework,Version=v10.0" were not found. To resolve this, install the SDK or Targeting Pack for this framework version or retarget your application to a version of the framework for which you have the SDK or Targeting Pack installed. Note that assemblies will be resolved from the Global Assembly Cache (GAC) and will be used in place of reference assemblies. Therefore your assembly may not be correctly targeted for the framework you intend.`
`1>Done building project "Opc.Ua.Core.csproj" -- FAILED.`

Where or how the TargetFrameworkIdentifier or TargetFrameworkVersion can be specified?

Greetings
Juergen

jh-isw, 2017-03-23:
Hello Juergen,
did Visual Studio 2017 ask for a unidirectional upgrade?
I had to install Windows 10 SDK to open UA-NetStandard Solution.
Hope this helps.

Greetings,
Jan

JuergenKosel, 2017-03-23:
Hello Jan,

> did Visual Studio 2017 ask for a unidirectional upgrade?

Yes, it has successfully done to unidirectional upgrade.

> I had to install Windows 10 SDK to open UA-NetStandard Solution.

The Windows 10 SDK versions 10.0.10586.0 and 10.0.14393.0 were already installed.
Now I have additionally installed 10.0.10240.0. Because it was the only not installed SDK in the list of the Visual Studio installer. But the build still fails, with the same output as in the first message above.

Greetings
Juergen

JuergenKosel, 2017-03-23:
I have pushed the generated files of the unidirectional upgrade to branch VisualStudio2017 in my fork of this repositry:
https://github.com/JuergenKosel/UA-.NETStandardLibrary/tree/VisualStudio2017

marcschier, 2017-03-23:
Hi @JuergenKosel, @jh-isw, as you already saw in the last pull request, we are working on this issue :-).  

In the meantime, if you have VS 2015 installed, you can add the following to the global.json file in the root folder:
`"sdk": {
     "version": "1.0.0-preview2-003131"
   }
`
This should enable you to build the UA-NetStandard.sln with VS 2015.

marcschier, 2017-03-23:
The changes are in.  Closing until further notice.

closed 2017-03-23 by marcschier