OPCFoundation/UA-.NETStandard #152: NullException in Opc.Ua.Client.Controls

aderr1, 2017-04-03:
In Session Session Create( 
            ApplicationConfiguration configuration,
            ConfiguredEndpoint       endpoint,
            bool                     updateBeforeConnect,
            bool                     checkDomain,
            string                   sessionName,
            uint                     sessionTimeout,
            IUserIdentity            identity,
            IList<string>            preferredLocales)
on Line 806 
sendCertificateChainOption = (configuration.Extensions.Find(e => e.Name.Equals("sendCertificateChain")).InnerText);

You'll get a NullReferenceException if no Extensions are specified. Its easy to fix this, so i hope you'll handle this with high priority.

mregen, 2017-04-26:
@aderr1 Hi, I think you are referring to an issue in this repo: https://github.com/OPCFoundation/UA-.NET. Both repos have a slightly diverged codebase, please move the issue to the other repo to get it fixed..
Thanks

aderr1, 2017-04-27:
OK

closed 2017-04-27 by aderr1