OPCFoundation/UA-.NETStandard #153: Library is compatible with the Mono Framework?

Dgn58, 2017-04-05:
Hello,

Is this Library compatible with the Mono Framework? 
I have to develop a PlugIn, which should send information to the cloud. The protocoll which I have to use should be the OPC UA and the PlugIn should written in C#. I need the Mono Framework, because the Software will run on a Linux machine.

I tried to compile or execute some of the Sample Applications, but without any success :( 

Can anyone please help?

BoeseB, 2017-04-05:
The Library targets .NetStandard 1.3 So according to [this table](https://github.com/dotnet/standard/blob/master/docs/versions.md) you should be able to use it with Mono 4.6

barnstee, 2017-04-05:
You don't need Mono to run this stack on Linux. .NetCore also support Linux:
https://www.microsoft.com/net/core#linuxredhat
The NetCoreConsoleClient and Server from the samples directory (https://github.com/OPCFoundation/UA-.NETStandardLibrary/tree/master/SampleApplications/Samples) therefore run on Linux.

Dgn58, 2017-04-05:
First of all I thank you very much for the quick help, but my big prolem is that I have to use Mono.
The rest of the Software is working on mono. 
Is there any way to use the Library on mono?

barnstee, 2017-04-12:
We haven't tried it. Since Mono is a clone of an older .Net API, there might be incompatibilities, so you will have to try it out yourself...

closed 2017-04-12 by barnstee