OPCFoundation/UA-.NETStandard #156: How to build on CentOS7

nnnntosh, 2017-04-11:
### question

Hi all, 
I tried to build on CentOS 7 with reference to "How to ..." in README.md but failed.
I am sorry, I don't know much about netstandard build system.
Please tell me what to do.

Write down the command I executed below.

```
# git clone https://github.com/OPCFoundation/UA-.NETStandardLibrary.git
# cd UA-.NETStandardLibrary
# dotnet restore UA-NetStandard_2017.sln

Welcome to .NET Core!
---------------------
Learn more about .NET Core @ https://aka.ms/dotnet-docs. Use dotnet --help to see available commands or go to https://aka.ms/dotnet-cli-docs.

Telemetry
--------------
The .NET Core tools collect usage data in order to improve your experience. The data is anonymous and does not include command-line arguments. The data is collected by Microsoft and shared with the community.
You can opt out of telemetry by setting a DOTNET_CLI_TELEMETRY_OPTOUT environment variable to 1 using your favorite shell.
You can read more about .NET Core tools telemetry @ https://aka.ms/dotnet-cli-telemetry.

Configuring...
-------------------
A command is running to initially populate your local package cache, to improve restore speed and enable offline access. This command will take up to a minute to complete and will only happen once.
Decompressing 100% 6027 ms
Expanding 100% 38430 ms
/root/.dotnet/sdk/1.0.1/Sdks/Microsoft.NET.Sdk/Sdk/Sdk.targets(40,3): error MSB4019: The imported project "/root/.dotnet/sdk/1.0.1/Microsoft/WindowsXaml/v/Microsoft.Windows.UI.Xaml.CSharp.targets" was not found. Confirm that the path in the <Import> declaration is correct, and that the file exists on disk. [/root/UA-.NETStandardLibrary/SampleApplications/SDK/Opc.Ua.Client/Opc.Ua.Client.csproj]
/root/.dotnet/sdk/1.0.1/Sdks/Microsoft.NET.Sdk/Sdk/Sdk.targets(40,3): error MSB4019: The imported project "/root/.dotnet/sdk/1.0.1/Microsoft/WindowsXaml/v/Microsoft.Windows.UI.Xaml.CSharp.targets" was not found. Confirm that the path in the <Import> declaration is correct, and that the file exists on disk. [/root/UA-.NETStandardLibrary/Stack/Opc.Ua.Core/Opc.Ua.Core.csproj]
/root/.dotnet/sdk/1.0.1/Sdks/Microsoft.NET.Sdk/Sdk/Sdk.targets(40,3): error MSB4019: The imported project "/root/.dotnet/sdk/1.0.1/Microsoft/WindowsXaml/v/Microsoft.Windows.UI.Xaml.CSharp.targets" was not found. Confirm that the path in the <Import> declaration is correct, and that the file exists on disk. [/root/UA-.NETStandardLibrary/SampleApplications/SDK/Opc.Ua.Configuration/Opc.Ua.Configuration.csproj]
/root/.dotnet/sdk/1.0.1/Sdks/Microsoft.NET.Sdk/Sdk/Sdk.targets(40,3): error MSB4019: The imported project "/root/.dotnet/sdk/1.0.1/Microsoft/WindowsXaml/v/Microsoft.Windows.UI.Xaml.CSharp.targets" was not found. Confirm that the path in the <Import> declaration is correct, and that the file exists on disk. [/root/UA-.NETStandardLibrary/SampleApplications/SDK/Opc.Ua.Server/Opc.Ua.Server.csproj]
/root/.dotnet/sdk/1.0.1/Sdks/Microsoft.NET.Sdk/Sdk/Sdk.targets(40,3): error MSB4019: The imported project "/root/.dotnet/sdk/1.0.1/Microsoft/WindowsXaml/v/Microsoft.Windows.UI.Xaml.CSharp.targets" was not found. Confirm that the path in the <Import> declaration is correct, and that the file exists on disk. [/root/UA-.NETStandardLibrary/SampleApplications/Samples/Opc.Ua.Sample/Opc.Ua.Sample.csproj]
  Restoring packages for /root/UA-.NETStandardLibrary/SampleApplications/Samples/NetCoreConsoleClient/NetCoreConsoleClient.csproj...
/root/.dotnet/sdk/1.0.1/NuGet.targets(97,5): error : Unable to resolve '/root/UA-.NETStandardLibrary/SampleApplications/SDK/Opc.Ua.Client/Opc.Ua.Client.csproj' for '.NETCoreApp,Version=v1.0'. [/root/UA-.NETStandardLibrary/UA-NetStandard_2017.sln]
/root/.dotnet/sdk/1.0.1/NuGet.targets(97,5): error : Unable to resolve '/root/UA-.NETStandardLibrary/Stack/Opc.Ua.Core/Opc.Ua.Core.csproj' for '.NETCoreApp,Version=v1.0'. [/root/UA-.NETStandardLibrary/UA-NetStandard_2017.sln]
  Generating MSBuild file /root/UA-.NETStandardLibrary/SampleApplications/Samples/NetCoreConsoleClient/obj/NetCoreConsoleClient.csproj.nuget.g.props.
  Generating MSBuild file /root/UA-.NETStandardLibrary/SampleApplications/Samples/NetCoreConsoleClient/obj/NetCoreConsoleClient.csproj.nuget.g.targets.
  Writing lock file to disk. Path: /root/UA-.NETStandardLibrary/SampleApplications/Samples/NetCoreConsoleClient/obj/project.assets.json
  Restore failed in 498.07 ms for /root/UA-.NETStandardLibrary/SampleApplications/Samples/NetCoreConsoleClient/NetCoreConsoleClient.csproj.
  Restoring packages for /root/UA-.NETStandardLibrary/SampleApplications/Samples/NetCoreConsoleServer/NetCoreConsoleServer.csproj...
/root/.dotnet/sdk/1.0.1/NuGet.targets(97,5): error : Unable to resolve '/root/UA-.NETStandardLibrary/SampleApplications/SDK/Opc.Ua.Configuration/Opc.Ua.Configuration.csproj' for '.NETCoreApp,Version=v1.0'. [/root/UA-.NETStandardLibrary/UA-NetStandard_2017.sln]
/root/.dotnet/sdk/1.0.1/NuGet.targets(97,5): error : Unable to resolve '/root/UA-.NETStandardLibrary/Stack/Opc.Ua.Core/Opc.Ua.Core.csproj' for '.NETCoreApp,Version=v1.0'. [/root/UA-.NETStandardLibrary/UA-NetStandard_2017.sln]
/root/.dotnet/sdk/1.0.1/NuGet.targets(97,5): error : Unable to resolve '/root/UA-.NETStandardLibrary/SampleApplications/Samples/Opc.Ua.Sample/Opc.Ua.Sample.csproj' for '.NETCoreApp,Version=v1.0'. [/root/UA-.NETStandardLibrary/UA-NetStandard_2017.sln]
/root/.dotnet/sdk/1.0.1/NuGet.targets(97,5): error : Unable to resolve '/root/UA-.NETStandardLibrary/SampleApplications/SDK/Opc.Ua.Server/Opc.Ua.Server.csproj' for '.NETCoreApp,Version=v1.0'. [/root/UA-.NETStandardLibrary/UA-NetStandard_2017.sln]
  Generating MSBuild file /root/UA-.NETStandardLibrary/SampleApplications/Samples/NetCoreConsoleServer/obj/NetCoreConsoleServer.csproj.nuget.g.props.
  Generating MSBuild file /root/UA-.NETStandardLibrary/SampleApplications/Samples/NetCoreConsoleServer/obj/NetCoreConsoleServer.csproj.nuget.g.targets.
  Writing lock file to disk. Path: /root/UA-.NETStandardLibrary/SampleApplications/Samples/NetCoreConsoleServer/obj/project.assets.json
  Restore failed in 346.63 ms for /root/UA-.NETStandardLibrary/SampleApplications/Samples/NetCoreConsoleServer/NetCoreConsoleServer.csproj.
  Restoring packages for /root/UA-.NETStandardLibrary/SampleApplications/Workshop/Aggregation/ConsoleAggregationServer/ConsoleAggregationServer.csproj...
/root/.dotnet/sdk/1.0.1/NuGet.targets(97,5): error : Unable to resolve '/root/UA-.NETStandardLibrary/SampleApplications/SDK/Opc.Ua.Configuration/Opc.Ua.Configuration.csproj' for '.NETCoreApp,Version=v1.0'. [/root/UA-.NETStandardLibrary/UA-NetStandard_2017.sln]
/root/.dotnet/sdk/1.0.1/NuGet.targets(97,5): error : Unable to resolve '/root/UA-.NETStandardLibrary/Stack/Opc.Ua.Core/Opc.Ua.Core.csproj' for '.NETCoreApp,Version=v1.0'. [/root/UA-.NETStandardLibrary/UA-NetStandard_2017.sln]
/root/.dotnet/sdk/1.0.1/NuGet.targets(97,5): error : Unable to resolve '/root/UA-.NETStandardLibrary/SampleApplications/SDK/Opc.Ua.Server/Opc.Ua.Server.csproj' for '.NETCoreApp,Version=v1.0'. [/root/UA-.NETStandardLibrary/UA-NetStandard_2017.sln]
/root/.dotnet/sdk/1.0.1/NuGet.targets(97,5): error : Unable to resolve '/root/UA-.NETStandardLibrary/SampleApplications/SDK/Opc.Ua.Client/Opc.Ua.Client.csproj' for '.NETCoreApp,Version=v1.0'. [/root/UA-.NETStandardLibrary/UA-NetStandard_2017.sln]
  Generating MSBuild file /root/UA-.NETStandardLibrary/SampleApplications/Workshop/Aggregation/ConsoleAggregationServer/obj/ConsoleAggregationServer.csproj.nuget.g.props.
  Generating MSBuild file /root/UA-.NETStandardLibrary/SampleApplications/Workshop/Aggregation/ConsoleAggregationServer/obj/ConsoleAggregationServer.csproj.nuget.g.targets.
  Writing lock file to disk. Path: /root/UA-.NETStandardLibrary/SampleApplications/Workshop/Aggregation/ConsoleAggregationServer/obj/project.assets.json
  Restore failed in 386.27 ms for /root/UA-.NETStandardLibrary/SampleApplications/Workshop/Aggregation/ConsoleAggregationServer/ConsoleAggregationServer.csproj.
  Restoring packages for /root/UA-.NETStandardLibrary/SampleApplications/Workshop/Reference/ConsoleReferenceServer/ConsoleReferenceServer.csproj...
/root/.dotnet/sdk/1.0.1/NuGet.targets(97,5): error : Unable to resolve '/root/UA-.NETStandardLibrary/SampleApplications/SDK/Opc.Ua.Configuration/Opc.Ua.Configuration.csproj' for '.NETCoreApp,Version=v1.0'. [/root/UA-.NETStandardLibrary/UA-NetStandard_2017.sln]
[... cut ...]
[... 19 lines cut ...]
  NuGet Config files used:
      /root/UA-.NETStandardLibrary/NuGet.Config
      /root/.nuget/NuGet/NuGet.Config

  Feeds used:
      https://api.nuget.org/v3/index.json

```

### Environment data
```
# dotnet --info
.NET Command Line Tools (1.0.1)

Product Information:
 Version:            1.0.1
 Commit SHA-1 hash:  005db40cd1

Runtime Environment:
 OS Name:     centos
 OS Version:  7
 OS Platform: Linux
 RID:         centos.7-x64
 Base Path:   /root/.dotnet/sdk/1.0.1
```

barnstee, 2017-04-12:
Did you follow the instructions here: https://www.microsoft.com/net/core#linuxcentos

nnnntosh, 2017-04-13:
Hi, barnstee,
Thank you for answer.

I tried it the way, but it is the same result.
"Sample Hello World application" could be built and executed.

barnstee, 2017-04-13:
From the error log above, it looks like your .NetCore insallation has failed. There are a number of issues on stackoverflow.com on .NetCore on CerntOS, please search and see if that helps you.

barnstee, 2017-05-08:
Is this issue still active?

nnnntosh, 2017-05-08:
My problem is not solved, but please close this issue.

closed 2017-05-08 by barnstee