OPCFoundation/UA-.NETStandard #158: Small Bug in SessionReconnectHandler of Opc.Ua.Client?

UwePhilippi, 2017-04-11:
I encountered an unexpected behavior when implementing a session reconnect functionality in an own net4 based client. 

In `private void OnReconnect(object state)` of SessionReconnectHandler the reconnection timer will be instantiated again even if reconnection was successful. When using the net4 based sample UA based client everything works as expected but this might be because of a race condition based on the InvokeRequired pattern used in `private void StandardClient_Server_ReconnectComplete(object sender, EventArgs e)` of ClientForm.cs.

In my case the timer stays alive because and cyclic try to reconnect the already reconnected session.

Solution: Add a return after m_callback(this,null);

```
                    // notify the caller.
                    m_callback(this, null);
                    return;
```

Thank you

Uwe

mregen, 2017-04-26:
Thanks, good find! I'll add your fix to my next update.

barnstee, 2017-05-18:
Leave a comment

closed 2017-05-18 by barnstee