OPCFoundation/UA-.NETStandard #159: Can anybody give an example of how to set a new value to an item?

mahaben, 2017-04-11:


UwePhilippi, 2017-04-12:
Hi mahaben,

here my client implementation of a write call utilizing the Opc.Ua.Client of the lib.
There is not only one truth how to implement things, hope this helps a bit.

```
        private void Write(List<IOpcUaHostingObject> WriteList)
        {
            StatusCodeCollection results = null;
            DiagnosticInfoCollection diagnosticInfos = null;
        
            try
            {  
                if (WriteList == null || WriteList.Count == 0) { return; }
                // build list of values to write.
                WriteValueCollection nodesToWrite = new WriteValueCollection();

                WriteList?.ForEach((Element) =>
                {
                    DataValue DataValueToWrite = Element.DataValue;
                    DataValueToWrite.ServerTimestamp = DateTime.Now;
                    nodesToWrite.Add(new WriteValue()
                    {
                        NodeId = NodeId.Parse("ns=" + Element.NamespaceIndex.ToString() + ";" +  Element.Identifier),
                        AttributeId = Attributes.Value,
                        Value = DataValueToWrite
                    });
                    
                });

                MySession.Write(
                    null,
                    nodesToWrite,
                    out results,
                    out diagnosticInfos);

                ClientBase.ValidateResponse(results, nodesToWrite);
                ClientBase.ValidateDiagnosticInfos(diagnosticInfos, nodesToWrite);

                results.ForEach((Element) =>
                {
                    // check results
                    if (StatusCode.IsBad(Element))
                    {
                        Utils.Trace(Utils.TraceMasks.All, "Message was : " + Element.ToString());
                    }
                });
            }
            catch (Exception ex)
            {
                Utils.Trace(Utils.TraceMasks.Error, "Exception in " + nameof(Write) + " with message: " + ex.Message);
            }
        }
```

Best,
Uwe

mahaben, 2017-04-12:
Works like a charm. Thank you very much @UwePhilippi

closed 2017-04-12 by barnstee