OPCFoundation/UA-.NETStandard #160: NetCoreConsoleServer and register server in discovery server

eoursel, 2017-04-14:
Hi, 

I noticed that the Registration of the NetCoreConsoleServer into the discovery server does not exist in Opc.Ua.SampleServer.Config.xml compared to the Net 4.5 sample server. I would like to know if it is just a missing section in the config file or if it is something more complex to add in the Netcore version.  Did you already test the discovery registration with the NetCore version? 

Thanks...

barnstee, 2017-04-18:
It's just missing. Registration should work identially acorss all frameworks, but we haven't tested .Net Core specifically. Please try it.

barnstee, 2017-05-08:
@eoursel, did you get a chance to try this out?

closed 2017-05-18 by barnstee

marcschier, 2017-06-20:
Fixed in #173