OPCFoundation/UA-.NETStandard #162: OPC client doesn't show updated/changed values from my server

roob82, 2017-04-21:
Hi!
I have been working with my OPC UA implementation for a OPC Server. I’m switching an old OPC Ua stack (about 8 years old) to the latest .NET in C#.
I got most things to work now with history etc.
I have replaced the obsolete CustomNodeManager with the new one CustomNodeManager2 in my application, where I have an own Nodemanager that inherits from the custom one.
But one thing that doesn’t seem to work. When I write to a tag in Ua Expert, it takes the value and I get the new value stored on my device. But I can not see any change in OPC Expert, It just shows my old value.
If i remove the tag and drag it in again I can see my new value. In the old CustomNodeManager this seemed to be handled by some polling methods (CreateSampledItem etc). This methods does not seem to exist anymore in the new CustomNodeManager2.
So I don’t really know how to get my values to be auto updated. Happy for some advice.
BR
Robert

barnstee, 2017-07-14:
Please contact Unified Automation for support on UA Expert.

closed 2017-07-14 by barnstee

mregen, 2017-07-14:
@roob82 In CustomNodeManager2 ClearChangeMasks must be called after a value change. Maybe thats missing in your case.