OPCFoundation/UA-.NETStandard #165: How to send more then one value in one subscription in Publisher(Azure )

tejasklodha, 2017-05-13:
Dear Sir,

I am using OPC Publisher.Net and able to send one value to azure IOT and Now I want to send multiple value in one subscription, Is it possible?

**Why I emphasis to multiple value in one subscription(Or Multiple value Publish at same time) :** 
Consider below example: 
_My OPC UA Server generate 40 variable in real time and if I subscribe 40 variable and publish each of variable then its generate 40 events in azure.... If I publish all 40 variables in one subscription then in azure generate 1 event...
and we already know that azure pay what I used means Azure IOT Hub charge on base of event generated._

**If it is not possible then How can I multiple subscription in real time by programming?
can you demonstrate ?** 
 
My purpose is simple that "Want to send multiple selected variable data to azure by programming"

**Note** : I know that we are not send too many variable in one time because I know that size lime on message(MESSAGE METER SIZE) is 4 KB

Thanks,

barnstee, 2017-05-16:
This should be possible by:
1. Collecting MonitoredItems in the Publisher (e.g. in a queue)
2. In regular intervals (e.g. every second or when a 4K message chunk is full), send the messages to IoTHub

Do you want try this out and send a pull request?

barnstee, 2017-07-14:
Closing as this should be implemented in the OPC Publisher.

closed 2017-07-14 by barnstee