OPCFoundation/UA-.NETStandard #169: UWP certificates - unable to locate Directory store - not in package localstate on Win10 IoT

JohnDrinkall, 2017-05-18:
I've just started looking at this, and while I have a sample UWP server up and running [on Windows 10 IoT], when a client connects from a different machine - I get the untrusted cert error.

Try as I might, I can't find where the certificates have been written to. I used nuget to download 0.2.4 of the OPC UA SDK. I've also tried referencing the projects in the server to see if I could step into the code, but the UWP app doesn't build in that scenario.

Any idea what I've missed. I saw that older versions didn't handle certificate generation, but that seems to have been resolved.

Thanks

J

barnstee, 2017-05-18:
It's in C:\Users\<username>\AppData\Local\Packages\<app guid>\LocalState\OPC Foundation\CertificateStores\UA Applications\certs

closed 2017-05-18 by barnstee

JohnDrinkall, 2017-05-18:
Thanks, I checked again, but nothing there. Behaviour seems different on PC compared to device. On the device I discover the services - but the connection isn't trusted. Running a UWP on desktop, Bad connection. I've used procmon to see if it writes any certificate files and it doesn't. 

The sample .net core works -- within a UWP and the nuget packages from microsoft it doesn't.

The readme.md file mentions building nuget packages from the build output and then including them into the UWP -- I'm assuming that since I couldn't simply reference the projects, including them this way gets around that. 

1) Are there any specific capabilities that I need to define in order for the UWP app to work?
2) I'll try and build the nuget packages from source and see if that makes a difference, and if I can work out how to step into the libraries debugging from a UWP app.

Cheers

J