OPCFoundation/UA-.NETStandard #174: The Application URI could not be read from the certificate. In new build

PedroFloresGuillen, 2017-06-05:
I cloned UA-.NETStandarLibrary, deleted the certs from machinedefault, done the dotnet restore, and tried the console sample server, getting this error.

Anyone can give me any idea about this problem? my own research has proven futile for the moment.

PedroFloresGuillen, 2017-06-08:
Windows 10/Visual studio 2015
[UA Core Sample Server [FF77EC3B4AFCB6C05CFAC4BF1CEDFC175F228F07].zip](https://github.com/OPCFoundation/UA-.NETStandardLibrary/files/1060742/UA.Core.Sample.Server.FF77EC3B4AFCB6C05CFAC4BF1CEDFC175F228F07.zip)

mregen, 2017-06-08:
Hi @PedroFloresGuillen, the cert looks good. Do you hit the error everytime during startup or did you just hit it once? 
There was one major change recently, application certs and private keys are now stored by default in the cert store. Previously they were stored in the filesystem. Then we added more fields into the SubjectName.
Please check if deleting all certs from the windows cert store CurrentUser\UA_MachineDefault fixes the issue.
Thanks,
Martin

PedroFloresGuillen, 2017-06-08:
Everytime I do the "dotnet run" command, this error appears, but if I press just enter the server will work as intented. Certs in the MachineDefault have been already deleted.

mregen, 2017-06-27:
@PedroFloresGuillen Sorry but I cannot repro the issue.

barnstee, 2017-07-14:
Closing as non repro, please reopen if still an issue.

closed 2017-07-14 by barnstee