OPCFoundation/UA-.NETStandard #175: Running a server without ApplicationCertificate

aanodin, 2017-06-05:
Hello!
I have a ridiculous SCADA system which seems to not support working with certificates at all. I suggested it because this SCADA could connect to a server based on open62541 without any security settings, but couldn't even get a list of nodes from a one built on UA-.NETStandardLibrary.  
So, can I somehow run a server without any security and ApplicationCertificate section at all? My example server is similar to SampleApplications/Samples/NetCoreConsoleServer and uses almost the same config. I need it to prove my suggestion.
I haven't found any information about it on google, so if anybody knows how to switch certificates off, please tell me.
Thanks a lot.

PedroFloresGuillen, 2017-06-05:
You can disable certificates in a server by making it accept everything (right now im working this way, i do not need to use any certificate to brownse my server)

aanodin, 2017-06-05:
If you mean something like this, I tried the following:
`<SecurityConfiguration>`
`  <AutoAcceptUntrustedCertificates>true</AutoAcceptUntrustedCertificates>`
`</SecurityConfiguration>`
but it raises an error about ApplicationCertificate node must be present (don't remember the exact message, but I can check).

Or you meant other setting?

PedroFloresGuillen, 2017-06-05:
No, I mean exactly that.

aanodin, 2017-06-05:
Could you please share the config that works for you? I can't make it work with such SecurityConfiguration, maybe the problem is somewhere else in my config.

PedroFloresGuillen, 2017-06-05:
My configuration is basically the demo server with the True, so that's not it. Maybe you have to send a certificate, even if you trust anyone?

aanodin, 2017-06-05:
That was an original question: could it NOT send a certificate to clients? When I connect using UaExpert it shows that the server sends it even when the setting set to true. When a server built on another OPC UA library (open62541) doesn't do it.

barnstee, 2017-06-06:
A few things:
1. If open62541 never sends a certificate, it uses the OPC UA nano server profile, which doesn't support any security. This is NOT RECOMMENDED for a production system and used at universities for simple tests only.
2. The use of open62541 for industry or production systems is NOT RECOMMENDED as it has not gone through OPC Foundation certification testing.
3. The setting AutoAcceptUntrustedCertificates is a server setting and ignored by clients. Again, this setting is NOT RECOMMENDED for industry or production systems.

barnstee, 2017-06-06:
Your "ridiculous SCADA system" badly needs to support security. Not supporting security is a complete no-go for production/industry systems. For testing purposes **only**, you can disable security by enabling "security mode none". Look for `<SecurityMode>None_1</SecurityMode>` in the .NetCoreConsoleServer config XML.

closed 2017-06-06 by barnstee