OPCFoundation/UA-.NETStandard #176: Pub/Sub specification already available

rafaelsf80, 2017-06-06:
Dear all,

Look like the [pubsub spec](https://opcfoundation.org/developer-tools/specifications-unified-architecture/part-14-pubsub) is available since some few months.
Is there any plans to implement this functionality in the sample codes ?

Thanks a lot in advance,
Rafa

dk2129, 2017-06-06:
The OPC Foundation published a Release Candidate (RC). However, the working group still validates the specification and already made several changes. There will likely be another RC before the Release which is targeted at Q4/2017.

dk2129, 2017-06-06:
The OPC Foundation plans to publish sample code for PubSub.

rafaelsf80, 2017-06-06:
Thanks for the update. 
Does the OPC Foundation have any ETA for the sample code for PubSub ?

mpostol, 2017-06-07:
Some example code you can find at: 

[Networking of SemanticData Library](https://github.com/mpostol/OPC-UA-OOI/tree/master/Networking/SemanticData)

The work is underway because the specification is unstable, but there is working example [UA Data Example Application](https://github.com/mpostol/OPC-UA-OOI/tree/master/Networking/ReferenceApplication) that is a proof of concept.

Unfortunately the release is  targeted at Q4/2017 as @K47S  said, so now we must work with what we have.

FYI: there is work targeting [Unsolicited Communication over MQTT](https://github.com/mpostol/OPC-UA-OOI/blob/master/Networking/SemanticData/README_MQTTMapping.md).

closed 2017-07-14 by barnstee

Shibynator, 2018-05-28:
How about today?

mpostol, 2018-05-28:
The specification has been released and I am working on implementation. Now priorities are: 
- mobile devices, namely the code will be ported to  .NETStandard
- AMQP transport
- Integration with Azure (IoT Hub,)
I have created an ebook at  https://mpostol.gitbooks.io/object-oriented-internet/content/Networking/

Unfortunately, the PubSub has the encoding and only encoding in common with OPC UA, therefore, PubSub should has an independent repo to be managed.

Shibynator, 2018-05-31:
Ok thx for the information. I am looking forward to the Pub/Sub implementation.:-)

mpostol, 2018-05-31:
@Shibynator  you are welcome to watch https://github.com/mpostol/OPC-UA-OOI/tree/master/Networking.
The specification has been released recently and there is a bunch of issues. The implementation is not an easy task and the specification stability is required for someone who is just an observer from a distance. If you are interested just join the project.