OPCFoundation/UA-.NETStandard #179: Message Sharing

aqib0080, 2017-06-13:
I am using this project in an IOT API. I want to send and receive messages between server and client. Please tell me how can I do this.
Rrgards

K47S, 2017-06-14:
Please have a look at the samples and the documentation [http://opcfoundation.github.io/UA-.NETStandardLibrary/](http://opcfoundation.github.io/UA-.NETStandardLibrary/)

aqib0080, 2017-06-21:
Dear
Can you please tell me a method to send and receive data using NetCoreConsoleClient and NetCoreConsoleSample. How can I make run time changes. Please donot refer the documentation. 
Regards

barnstee, 2017-07-14:
This is already implemented in the netcoreconsoleclient. Have a look at the code where subscriptions are added. Also, please read up on OPC UA communication patterns so you understand the concepts of Client-Server communication in OPC UA.

closed 2017-07-14 by barnstee