OPCFoundation/UA-.NETStandard #180: More information about Console Samples restoring

JayeshThamke, 2017-06-20:
Hello,
This may be a small doubt. 
I did not understand the Prerequisite section in README.md regarding restoring. 
How can I get the root path of local repository?
I have considered 
`F:\gitHubRepo\opcF\UA-.NETStandardLibrary `
as my root path of local repository. And I tried to `dotnet restore` but it gave me MSBUILD error 

MSB1011: Specify which project or solution file to use because this folder contains more than one project or solution file.
I am pretty new to it. Could you please assist me in this case? Thanks,

mregen, 2017-06-27:
@JayeshThamke Hi, sorry for the inconvenience, but latest .Net versions (e.g. 1.0.4 on Linux) changed behaviour of dotnet and moved to new project files (xproj-->csproj). We are working on an upgrade to fully support VS2017 and 1.0.4. you can only try to run: dotnet restore <SolutionFile> and see if it works. Or move to an older version of dotnet (preview on Linux or VS2015 on windows).

mregen, 2017-07-04:
@JayeshThamke Hi, please give a try. With the latest pull request all should work again.

JayeshThamke, 2017-07-06:
hi @mregen 
Referring to the prerequisite in documentation - 

> Prerequisites

Once the dotnet command is available, navigate to the root folder in your local copy of the repository and execute dotnet restore. This command calls into NuGet to restore the tree of dependencies.

- is not clear to me. I did not understand the root folder mentioned above. Exactly what folder (location) ?

I have cloned the repository in C:\github_repos\opcua 
After the cloning is done I navigated to C:\github_repos\opcua\UA-.NETStandardLibrary folder and run using - dotnet run . 
The powershell return following text.

I am sure, I am missing some really basic setting. but don't know what.

`Windows PowerShell
Copyright (C) 2009 Microsoft Corporation. All rights reserved.

PS C:\Windows\system32> cd C:\github_repos\opcua
PS C:\github_repos\opcua> git clone https://github.com/OPCFoundation/UA-.NETStandardLibrary.git
Cloning into 'UA-.NETStandardLibrary'...
remote: Counting objects: 8701, done.
remote: Compressing objects: 100% (80/80), done.
remote: Total 8701 (delta 47), reused 72 (delta 29), pack-reused 8590
Receiving objects: 100% (8701/8701), 16.15 MiB | 4.19 MiB/s, done.
Resolving deltas: 100% (6002/6002), done.
PS C:\github_repos\opcua> cd .\UA-.NETStandardLibrary
PS C:\github_repos\opcua\UA-.NETStandardLibrary> dotnet

Microsoft .NET Core Shared Framework Host

  Version  : 1.1.0
  Build    : 928f77c4bc3f49d892459992fb6e1d5542cb5e86

Usage: dotnet [common-options] [[options] path-to-application]

Common Options:
  --help                           Display .NET Core Shared Framework Host help.
  --version                        Display .NET Core Shared Framework Host version.

Options:
  --fx-version <version>           Version of the installed Shared Framework to use to run the application.
  --additionalprobingpath <path>   Path containing probing policy and assemblies to probe for.

Path to Application:
  The path to a .NET Core managed application, dll or exe file to execute.

If you are debugging the Shared Framework Host, set 'COREHOST_TRACE' to '1' in your environment.

To get started on developing applications for .NET Core, install the SDK from:
  http://go.microsoft.com/fwlink/?LinkID=798306&clcid=0x409
PS C:\github_repos\opcua\UA-.NETStandardLibrary> dotnet run
Couldn't find a project to run. Ensure a project exists in  C:\github_repos\opcua\UA-.NETStandardLibrary.
Or pass the path to the project using --project
PS C:\github_repos\opcua\UA-.NETStandardLibrary>

`

mregen, 2017-07-07:
@JayeshThamke I need to double check if the documentation is still correct. But for command line runs first you run `dotnet restore` in the repo root (e.g. C:\github_repos\opcua\UA-.NETStandardLibrary). Then you cd into the folder of a Console app and run the sample using: `dotnet run`. Even better now there is full support for VS2017. Also make sure you have the latest .Net Core SDK 1.1 installed (coming with VS2017). .Net 2.0 is not yet supported.

barnstee, 2017-07-07:
I fixed this my moving the iop comm solution to the iot com folder. dotnet restore in the rrot of the repo should work now.

barnstee, 2017-07-14:
Closing, please reopen if still an issue.

closed 2017-07-14 by barnstee