OPCFoundation/UA-.NETStandard #181: Connect with ModBus TCP

MSIoT, 2017-06-21:
Does server able to connect to Modbus TCP and where to setup it up?

eoursel, 2017-06-21:
No. It does not. 

Nevertheless, i have written an extension to UA .NetStandardLibrary which implement a Modbus TCP Master. This is a proof of concept not a fully tested product and i didn’t decide yet if i will publish this work. Feel free to contact me directly on eric.oursel@outlook.com.  

Best regards, 

Eric OURSEL
OPC Foundation Technical Advisory Council

De : Microsoft IoT Innovation Center Team Projects
Envoyé le :mercredi 21 juin 2017 11:26
À : OPCFoundation/UA-.NETStandardLibrary
Cc : Subscribed
Objet :[OPCFoundation/UA-.NETStandardLibrary] Connect with ModBus TCP (#181)

Does server able to connect to Modbus TCP and where to setup it up?

barnstee, 2017-07-14:
There are plenty Modbus->OPCUA adapters out there, e.g. here: https://www.kepware.com/en-us/products/kepserverex/suites/modbus-suite/

closed 2017-07-14 by barnstee